BelfrySCAD/BOSL2 #1379: Libraries folder for OpenSCAD snap nightly installation

ifishtoo, 2024-02-15:
I am brand new to OpenSCAD.  I installed the snap nightly build in ubuntu.  When I tried to copy to the libraires folder I still got error messages.  I just happen to open the help and noticed library info. This revealed the correct path to copy the files too. _User Library Path: /home/<user>/snap/openscad-nightly/4175/.local/share/OpenSCAD/libraries._ Do you suppose you could add an appropriate note to the instructions like "check help/libraries info" for correct path ? The stock location provided for linux does not work for snap installations.

adrianVmariano, 2024-02-15:
A much better strategy is to pick a reasonable library path that you can find on your own.   I use:

`export OPENSCADPATH=$HOME/scad/lib`

adrianVmariano, 2024-02-18:
Fixed.

closed 2024-02-18 by adrianVmariano