BelfrySCAD/BOSL2 #1382: [BUG] offset_sweep() fails when passed a region

AUnicyclingProgrammer, 2024-02-18:
**Describe the bug**
Rounding module offset_sweep() fails when provided regions in BOSL2 version 2.0.716, but works as expected in BOSL2 version 2.0.652

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

rectPath = rect([40, 60]);
rectRegion = make_region(rectPath);
manualRectRegion = [rectPath];

echo("Conforming that rectPath is a path", is_path(rectPath));
echo("Conforming that rectRegion is a region", is_region(rectRegion));
echo("Conforming that manualRectRegion is a region", is_region(manualRectRegion));

// This command works in version 2.0.652 and version 2.0.716
*offset_sweep(rectPath, height = 20);

// This command works in version 2.0.652 but not in version 2.0.716
*offset_sweep(rectRegion, height = 20);

// This command works in version 2.0.652 but not in version 2.0.716
*offset_sweep(manualRectRegion, height = 20);
```

**Expected behavior**
Each of the three commands provided should produce a cube with the dimensions [40, 60, 20]. offset_sweep() began to throw the error after I updated versions of the BOSL2 library.

Error Message: (extra path info removed from the file path)
```
ERROR: Assertion 'is_region(region)' failed: 2 in file .../OpenSCAD/libraries/BOSL2/attachments.scad, line 2946
TRACE: called by 'attach_geom' in file .../OpenSCAD/libraries/BOSL2/attachments.scad, line 2597
TRACE: assignment to 'geom' in file .../OpenSCAD/libraries/BOSL2/attachments.scad, line 2596
TRACE: called by 'attachable' in file .../OpenSCAD/libraries/BOSL2/rounding.scad, line 1656
TRACE: called by 'if' in file .../OpenSCAD/libraries/BOSL2/rounding.scad, line 1654
TRACE: call of 'offset_sweep(path = [[[20, -30], [-20, -30], [-20, 30], [20, 30]]], height = 20, bottom = [], top = [], h = undef, l = undef, offset = "round", r = 0, steps = 16, quality = 1, check_valid = true, extra = 0, cut = undef, chamfer_width = undef, chamfer_height = undef, joint = undef, k = 0.75, angle = 45, convexity = 10, anchor = "origin", cp = "centroid", spin = 0, orient = [0, 0, 1], atype = "hull")' in file .../OpenSCAD/libraries/BOSL2/rounding.scad, line 1647
TRACE: called by 'offset_sweep' in file , line 16
Compiling design (CSG Products generation)...
Geometries in cache: 802
Geometry cache size in bytes: 1132360
```

**Screenshots**
Not applicable

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: [OpenSCAD](https://www.openscad.org/) version 2023.12.09 (git 3d094bf87)
- Other libraries used: None

adrianVmariano, 2024-02-18:
I'm finding actually the reverse behavior, that it fails when given a path and succeeds only when given a region.  (Note that offset_sweep doesn't really accept regions----only regions that have a single component.)  

The fix for the problem I'm seeing is to change the attachable call in module offset_sweep to read

```
        attachable(anchor,spin,orient,region=force_region(path),h=h,extent=atype=="hull",cp=cp){
```

(line 1656 of rounding.scad)  But if you're seeing a reversed problem, not sure what's going on there.

adrianVmariano, 2024-02-18:
I think this is fixed.  All three forms are working.  If it isn't, reopen the issue.

closed 2024-02-18 by adrianVmariano

AUnicyclingProgrammer, 2024-02-18:
Here is a screenshot of what OpenSCAD generates when I call it on my machine..

[image]

I don't know if it is relevant, but I'm using a Windows machine, and in my OpenSCAD preferences I have both fast-csg-safer and manifold enabled.
[image]

Just got your message, I will download the latest version of the library and try it. Thank you.

AUnicyclingProgrammer, 2024-02-18:
> I think this is fixed. All three forms are working. If it isn't, reopen the issue.

After downloading the latest version of the library all three commands now work as expected.
[image]

Thank you for resolving this issue so quickly.