BelfrySCAD/BOSL2 #1387: Expose unique $idx for each child created by grid_copies()

alastaira, 2024-02-19:
Unlike (all?) other distributors, grid_copies() currently doesn't expose an $idx index for each child.

**Describe the solution you'd like**
A unique identifier to be able to address each child object created by grid_copies, and to ensure feature parity and consistency with $idx exposed by other distributors.

**Describe alternatives you've considered**
grid_copies() does expose separate $row and $col, which can be used to derive a unique index for each child for simple aligned grids (i.e. $row*COLS + $col), but this fails when, e.g. rows have been staggered, or when the grid was created without specifying a known number of rows and columns (such as when using the "inside" region)

Thankyou!

adrianVmariano, 2024-02-19:
So you want a linear index in no particular order for the children?

revarbat, 2024-02-19:
`grid_copies()` has `$row=` and `$col=` instead of `$idx=`.  Do you still need `$idx=`?

revarbat, 2024-02-19:
`$col=` should be consistent in behavior even with staggering.  However, when using a shape mask, `$row=`/`$col=` pairs that would appear outside the mask shape just won't be used.

adrianVmariano, 2024-02-19:
He said that `$row` and `$col` aren't enough.  It's nontrivial to create an index...even within the code.

adrianVmariano, 2024-02-19:
The module and function version of grid_copies are copy-and-paste versions of each other.  Kind of ugly.    Should be consolidated....

adrianVmariano, 2024-02-20:
This is updated.

closed 2024-02-20 by adrianVmariano

alastaira, 2024-02-20:
Amazing, that is perfect. 
I must congratulate all the maintainers of this repository for being some of the most responsive developers I've ever dealt with (on Github, or anywhere else). Thankyou. x