BelfrySCAD/BOSL2 #1388: Feature Request : 2D version of bounding_box()

AUnicyclingProgrammer, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
I have a simple 2D region that I want to take the outline of. I was hoping the library had a 2D version of bounding_box() for either paths or regions, but I don't see any in the online documentation.

**Describe the solution you'd like**
A version of bounding_box() that outputs regions and paths, much like the relationship between the hull operation and hull2d_path() and hull_region().

**Describe alternatives you've considered**
I've considered using the hull operations or offset operations to determine the bounding box, but given that I need a rectangular bounding box I don't think either of those options would work.

In the end I decided to write a sample function that did what I wanted.

**Example Code**
```
include <BOSL2/std.scad>

xdistribute(spacing = 30) {
    union() {
        star = star(n = 5, id = 10, od = 20);
        starBoundingBox = bounding_box2d(star);
        
        stroke(star, closed = true);
        stroke(starBoundingBox, closed = true);
    }
    
    union() {
        blob = glued_circles(r = 5, spread = 10, tangent = 60);
        blobBoundingBox = bounding_box2d(blob);
        
        stroke(blob, closed = true);
        stroke(blobBoundingBox, closed = true);
    }
    
    union() {
        // Example provided in documentation
        supershape = supershape(step=0.5,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,d=20);
        supershapeBoundingBox = bounding_box2d(supershape);
        
        stroke(supershape, closed = true);
        stroke(supershapeBoundingBox, closed = true);
    }
}

// Proposed Function (may need optimization)
function bounding_box2d(path) =
    let(
        bounds = pointlist_bounds(path),
        
        xLen = abs(bounds[0].x - bounds[1].x),
        yLen = abs(bounds[0].y - bounds[1].y),

        center = mean(bounds),

        bounding_box = move(center, p = rect([xLen, yLen]))
    ) bounding_box;
//
```

**Screenshots**
[image]

**Additional context**
This type of function feels like something that would fit nicely in the BOSL2 ecosystem.

adrianVmariano, 2024-02-19:
I've never used bounding_box() and can't think of any application for the proposed module.  Why is it useful?

revarbat, 2024-02-19:
Do you actually need the 2D geometry of a bounding box, or do you need to know the minimum and maximum X, Y, and Z components of a Region that is given as a list of paths?

If the latter, you can use `pointlist_bounds(flatten(rgn))`

revarbat, 2024-02-19:
Ah, I see you just want the bounding box path.

```openscad
function bounding_rect(rgn) =
    let(
        pts = flatten(force_region(rgn)),
        b = pointlist_bounds(pts)
    ) [
        [b[1].x, b[0].y], b[0],
        [b[0].x, b[1].y], b[1]
    ];
```

AUnicyclingProgrammer, 2024-02-19:
Yes, that code would generate exactly what I was looking for.

adrianVmariano, 2024-02-19:
Why do you need a "negative" of a shape that you have represented as a path or region?  I don't understand what this enables you to do.

AUnicyclingProgrammer, 2024-02-20:
On this occasion I was trying to make a tabbed part with offset_sweep, but I kept running into assertion errors when check_valid was true and it created artifacts if I disabled check_valid. I was hoping that the artifacts were directional and would disappear if I differenced the tab with its negative, but the negative created the same artifacts. After a lot of troubleshooting, I finally determined that the problems were a result of creating arcs with too high of an $fn value. (In hindsight, that idea seems a lot worse than it did to me at the time.)

I understand that it doesn't make sense to add features unless they serve a purpose. I've only been able to come up with one valid use case thus far, being that it could be used to easily create frames around patterns, mimicking the way the ancient romans decorated the weight saving recesses in their ceilings.

[image]

adrianVmariano, 2024-02-20:
It doesn't sound like this is of general interest.  If it comes up again from other users we can reconsider.  

Note that for framed patterns actually you probably want all the frames the same size, so you have to pick the frame size and fit your decorations into the frame, like the middle one looks like it doesn't touch the frame edges.  So this particular method wouldn't be suited to the framed pattern problem.

closed 2024-02-20 by adrianVmariano

AUnicyclingProgrammer, 2024-02-21:
I agree with your decision, thank you for taking the time to consider this.