BelfrySCAD/BOSL2 #1392: Expose Anchors when using Attachable()

AUnicyclingProgrammer, 2024-02-23:
**Is your feature request related to a problem? Please describe.**
When designing, I typically use methods with descriptive names to encapsulate subcomponents, even if that's only a single object (like the output of a sweep or skin operation). I'd like a way to pass the anchors from a child object directly to the parent object so I don't have to manually re-create them anytime I want to give something a better name.

Below I have included a couple of screenshots of an impeller blade I made using the skin operation. I'd like to pass the anchors directly through so I don't have to re-create them by hand.

**Describe the solution you'd like**
I think a similar solution to expose_tags from this issue (https://github.com/BelfrySCAD/BOSL2/issues/1242) would work well.

Perhaps an expose_anchors option where the parent object automatically inherits all the anchors of the child object, but not the anchors of any subsequent child objects (similar to how color_this() behaves).

**Describe alternatives you've considered**
I've recreated the attachments by hand a few times, but depending on the geometry (like the output from a skin or path_sweep2d operation) it can be tedious to do.

I apologize if this is a redundant request and this type of behavior already exists. I'm still learning how to use the attachment system and wouldn't be surprised if I overlooked something in the documentation.

**Additional context**
[image]
[image]

adrianVmariano, 2024-02-23:
So you want the parent object to be able to do something like `child_anchor(TOP)` and do something with that?  

If that's what you want, that's impossible.  Information only flows down in OpenSCAD.  Parents have no way of extracting any information about their children.

adrianVmariano, 2024-02-23:
If you want behavior like

```
parent(){
   foo=child_function(.....);
   attachable(....,geom=get_geom(foo),...){
        make_geometry(foo);
       children();
}}
```

that could maybe do what you're after?

AUnicyclingProgrammer, 2024-02-23:
> So you want the parent object to be able to do something like `child_anchor(TOP)` and do something with that?

Yes, that is the kind of think I was thinking of doing. I forgot about that limitation with OpenSCAD.

Might is be possible to defer the transformation operations to the child object?  

> If you want behavior like
> 
> ```
> parent(){
>    foo=child_function(.....);
>    attachable(....,geom=get_geom(foo),...){
>         make_geometry(foo);
>        children();
> }}
> ```
> 
> that could maybe do what you're after?

That might do what I want, but I'm not sure what child_function() is. I'm assuming it's a VNF operation or something of a similar nature.

adrianVmariano, 2024-02-23:
That code was all notional.  child_function() is the function you write that produces anchor information that you want to make use of upstream and some representation you can use to create child: so it could return a vnf, or it could return some other description of the child.  The above is all possible already.  

I'm not sure what you mean about "defer transformation operations to the child".  You can always apply transformations to the child by just doing it.

AUnicyclingProgrammer, 2024-02-24:
> That code was all notional. child_function() is the function you write that produces anchor information that you want to make use of upstream and some representation you can use to create child: so [...]

Oh, that makes sense, thank you.

> I'm not sure what you mean about "defer transformation operations to the child". You can always apply transformations to the child by just doing it.

I'm not sure how to distill this idea into a few words, but I can walk though my thought process.

As I mentioned before, often I create modules that only contain one object, like the impeller vane I mentioned earlier. Since I use BOSL2 a lot now, typically that object already has anchors and supports the attachment system.

As I understand it, there appears to be a trade-off between making my code more readable or being able to use the attachment system. In essence I'm just renaming something, so this trade-off doesn't feel necessary.  (Going back to the impeller example, I'm essentially renaming a specific skin call "impellerBlade")

In this type of situation I think it may be possible to pass the attachment commands directly to the child module because the parent module doesn't actually create anything new.

This example doesn't actually accomplish anything, but hopefully it helps explain the idea better. I'm assuming this behavior is tied defer_attachments_to_child.

```
module Box(anchor, spin, orient) {
     attachable(anchor, spin, orient, defer_attachments_to_child = true) {
          cuboid(10, anchor = anchor, spin = spin, orient = orient);
          children();
     }
}
```

This code definitely won't do what I want it to, (direct pass-through probably isn't the correct way to go about this, nor do I know how make this support the position, align, or attach commands), but hopefully that helps clear things up a bit.

adrianVmariano, 2024-02-24:
wouldn't something like

```
Module Box(anchor, spin, orient) {
   cuboid(10, anchor=anchor, spin=spin,orient=orient) children();
}
```

do what you want?

AUnicyclingProgrammer, 2024-02-24:
Yes it does, I didn't realize it was that simple to implement. Thanks for your help.

Could you cover this in the [Making Attachables](https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Attachments#making-attachables) portion of the attachments tutorial? I checked that multiple times trying to figure this out.

adrianVmariano, 2024-02-26:
Added a section about this to the tutorial.

closed 2024-02-26 by adrianVmariano