BelfrySCAD/BOSL2 #1397: Partition Feature Request - Anchor Based Partitioning

AUnicyclingProgrammer, 2024-03-01:
**Is your feature request related to a problem? Please describe.**
Often when I'm working on a design I'll mask off part of it to view the internal geometry. The default planar cutting options are useful, but often I want to remove a smaller area, like a quarter of a piece.

**Describe the solution you'd like**
It would be useful to have a function that used the default anchors to partition off certain parts of the object relative to the origin.

For ease of use and implementation, it should only support one anchor location. The user shouldn't be able to input [FRONT, BACK+LEFT], just FRONT or BACK+LEFT.

**Example Code**
```
mask_off(BACK+RIGHT) cube(10); //Removes all material in the +X +Y region of the part

mask_off(RIGHT) cube(10); // Same as left_half();

mask_off(BACK+RIGHT+TOP) cube(10); // Removes all material in the +X +Y +Z region of the piece
```

adrianVmariano, 2024-03-01:
I'm not sure I understand what you want.  It sounds like you want the equivalent of

difference(){
    object();
    cuboid(100,anchor=FRONT+LEFT);
}

or similar.  I'm not sure that this is worth implementing.

AUnicyclingProgrammer, 2024-03-01:
I'm looking for something like this.
[image]

You're proposed solution is how I've created this cutout, but I mainly use this type of behavior for troubleshooting when I'm designing, and that implementation doesn't work how I'd like it to.

Typically when I'm designing I'll toggle between certain views by using a keyboard shortcut to comment or uncomment a line. With the standard partitions I can toggle between looking at half of the object or the entire object, but when I'm working with rotationally symmetrical objects I find it a lot more useful to only remove a quarter of an object.

adrianVmariano, 2024-03-02:
I don't understand what the problem is.  Comment out the cuboid line if you don't want it subtracted.

AUnicyclingProgrammer, 2024-03-02:
I suppose this is less of a necessary feature and more of a quality of life improvement. I felt it was worth proposing in case others also felt it would be useful.

I like to try and keep my code clean. With a one line solution I can simply remove that line when I'm done prototyping and move on. With my current implementation I have to remove multiple lines and fix the indentation. (I realize an auto-formatter would fix the indentations, but I'm not aware of any that would automatically remove the other redundant lines of code)

adrianVmariano, 2024-03-02:
I don't think there's a strong enough case for adding this to the library.  You can code it yourself if you want a one line solution for this.  Something like `module cubemask(dir,size) {difference(){cuboid(size,anchor=-dir);children();}}`

closed 2024-03-02 by adrianVmariano