BelfrySCAD/BOSL2 #1401: [BUG] Inconsistent result when using _root_radius: Too small in some cases.

ReallyCoolName, 2024-03-07:
Two gears, both with hidden teeth, same module number but different number of teeth overall
For one, `_root_radius` results in a radius that is just right (it fits in with the result of `spur_gear` with hidden teeth). For another, the radius is too small.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/gears.scad>
$fn = 100;
union() {
    spur_gear(mod = 5, teeth = 18, hide=14, thickness = 3);
    cylinder(r = _root_radius(mod = 5, teeth = 18),h = 3, center=true);
}

translate([20,0,5]) union() {
    spur_gear(mod = 5, teeth = 12, hide=8, thickness = 3);
    cylinder(r = _root_radius(mod = 5, teeth = 12),h = 3, center=true);
}
```

**Screenshots**
[image]

**Additional context**
- OpenSCAD Version: 2024.02.29
- Other libraries used: None

adrianVmariano, 2024-03-07:
Technically speaking a bug in _root_radius is impossible because this is an internal function.    Does it need to be part of the API?

Note that the documentation that you can read in the code, which happens to be correct, states:   Does not apply auto profile shifting.   Have you read the gears intro section at the top of gears.scad?   I'm wondering if some clarification is needed in that section.  You'll get the result you expect if you add the profile shift.  Or if you remove it from the gear.

ReallyCoolName, 2024-03-08:
I assumed profile shifting to be something else, just read about what it actually is, so I guess if that's all there is to it, that's my bad.

As for whether or not it needs to be part of the API - I find that I do need the root radius, but I'm just the one person. I could also use the formula (z-2.5)*mod that I found online, which gives me the same results.

in the meantime, taking `outer_radius - (addendum+dedendum)` solved the problem for me

closed 2024-03-08 by ReallyCoolName