BelfrySCAD/BOSL2 #1403: Partial Texturing of Objects

RAMilewski, 2024-03-08:
Currently texturing is applied over the entire surface of the object.  Setting tex_reps to [1,1] stretches the texture to fit.  This distorts surface textures created with img2scad.py.  

It would be useful to be able to specify that a texture be applied without stretching to fit.  Perhaps tex_reps[0,0] ?

This can be done now by leaving large margins in the .png file supplied to img2scad.py, however this results in painfully slow processing in most cases.

[image]

adrianVmariano, 2024-04-12:
It seems to me that processing to make this happen would be to **internally** add large margins to the texture data.  So it wouldn't help with the slow processing problem.   Could that be addressed by putting the texture only on a small section and unioning with the rest?

RAMilewski, 2024-05-14:
I think texturing a small section and unioning the rest is probably the right answer.  Having a heightfield replacement (textured cuboid top?) would enable texturing part of the flat surfaces of prismoids, and other irregular shapes with flat surfaces.

closed 2024-05-14 by RAMilewski