BelfrySCAD/BOSL2 #1404: [BUG] BOSL2 Version Not Incrementing

AUnicyclingProgrammer, 2024-03-09:
**Describe the bug**
I downloaded the latest version of the library today (3/9/2024), version 2.0.716.
This never version has more functionality than the previous version I downloaded on on 2/19/2024, that version had the same version number: 2.0.716.

**Expected behavior**
I expected the version number to change after downloaded the latest version of the library, even it if was only to version 2.0.717.

**Additional context**
This bug was mentioned in [Issue 1036](https://github.com/BelfrySCAD/BOSL2/issues/1036#issuecomment-1694124870)

- OpenSCAD Version: 2024.02.22 (git 2eb528b27)
- Other libraries used: None

adrianVmariano, 2024-03-10:
The automatic version number increment feature that was implemented turned out to cause a bunch of problems, so Revar backed it out.

AUnicyclingProgrammer, 2024-03-10:
Is there a proposed solution to this issue in the works? I'd imagine that having an up to date version number makes many troubleshooting tasks easier.

If there isn't a proposed solution in the works, perhaps you could create a new release every Sunday at 12:00 a.m. with a naming scheme such as Version 2.0_week# or Version 2.0.mm_dd_yy. (Assuming there is a way to auto-create releases using some form of GitHub automation)