BelfrySCAD/BOSL2 #1412: generic_bottle_cap module fails with variable "thread_od" not specified.

JemC42, 2024-03-31:
**Describe the bug**
Tried to use the generic_bottle_cap module but gives me an error
"WARNING variable thread_od not specified as a parameter"

when I remove the parameter it generates a model and adding "wall" as a parameter changes the model.
This is inconsistent with the documentation.

**Describe the bug**
A clear and concise description of what the bug is.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/bottlecaps.scad>
generic_bottle_cap(texture="ribbed",height=10,pitch=2.5,thread_od=15, neck_od=17);
```

**Expected behavior**
Expectation would be to specify outside diam of the cap .

**Screenshots**
n/a

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: none

adrianVmariano, 2024-04-01:
No, actually, thread_od would NOT be the outer diameter of the cap.  It would be the diameter to the bottom of the threads inside the cap.  It's not a very intuitive dimension on the cap.  It would be the outer diameter of the threads of the mating bottle neck---well, a little more to give clearance.  

The internal computation of the outer diameter is kind of baffling.  I don't know where it came from.  It says:

```
    threadOuterDTol = neck_od + 2*(thread_depth - 0.8) + 2 * tolerance;   // WTF; Engineered for consistency with old code, but 
    w = threadOuterDTol + 2 * wall;                                       // no clue why this was chosen
```

I could add direct support for thread_od which would add `2*tolerance` to it instead of the above mysterious computation.  

Why are you using generic bottle cap instead of one of the more specific ones?

adrianVmariano, 2024-04-01:
One observation is that if you specify neck_od and thread_od then thread_depth is determined.  In practice, threading on bottles is specified by giving a thread depth, not inferring it from ID and OD.

JemC42, 2024-04-01:
my use case was not strickly a bottle. Wanted a cap for an unusual thread and I thought the generic bottle cap might work for me.. Followed the documentation where thread_od is defined as a parameter (but thread_depth isn't)
It makes sense that the thread_od is not the outside diam of the cap.
Maybe not a bug but just an update to the wiki?
Thanks for the response.

adrianVmariano, 2024-04-01:
So you're trying to mate with an existing thread?  What would you see as the most natural or easiest way to define what you need?

I have to say that computation in there right now for computing the thread OD doesn't make much sense.   It looks like you specify a neck OD and a thread depth and then the actual thread depth is adjusted by (tolerance-0.4).  So if tolerance is smaller than 0.4 it will *decrease* the thread depth from what you requested.

adrianVmariano, 2024-04-02:
I overhauled generic_bottle_cap to stop using the weird 0.4 adjustment (which was burying the thread by 0.4 units into the side wall of the cap, so not only is the thread depth not what you expect, it doesn't have the correct form).  

I added support for thread_od and updated docs.  This is all in the PR.  

I also removed the weird random defaults that were undocumented.

closed 2024-04-02 by revarbat

JemC42, 2024-04-02:
Wll try out your changes. Thanks.