BelfrySCAD/BOSL2 #1415: [BUG] partition_cut_mask says it has a width parameter but it doesn't

zyphlar, 2024-04-03:
**Describe the bug**
partition_cut_mask(l=1,w=1,h=1) is valid according to the [documentation](https://github.com/BelfrySCAD/BOSL2/wiki/partitions.scad#module-partition_cut_mask) but isn't actually implemented in code; `w` isn't a parameter.

**Code To Reproduce Bug**
```
partition_cut_mask(l=1,w=1,h=1);
```

**Expected behavior**
I can adjust all dimensions of my dovetail without resorting to scale()

adrianVmariano, 2024-08-23:
This is just a documentation issue, right?  The parameter w does not exist and needs to be removed from the docs.  It isn't relevant to the cut mask.

zyphlar, 2024-08-23:
I think I expected to be able to scale my mask, hence the scale() mention. If the width parameter isn't supposed to be used for that, then my solution is proper and it's just a documentation issue

adrianVmariano, 2024-08-23:
But isn't the cut mask just the line between the two things?  Why would you want to scale that?  Scaling would make the gap non-uniform in thickness so I don't think that's correct.   I think if you want a larger dovetail  you increase cutsize and/or gap.  Revar responded to this issue by trying to completely rewrite the partitioning code...but that project is evidently stalled, so I thought maybe I'd try to at least fix the docs.

zyphlar, 2024-08-23:
Here's the implementation I used:

https://git.zyphon.com/will/doggie-door/src/branch/main/doggie-door.scad#L171

adrianVmariano, 2024-08-23:
I think instead of scale you should have done:

`partition_cut_mask(gap=0, cutsize=5, cutpath="dovetail", $slop=0.2);`

which maintains a uniform thickness.

adrianVmariano, 2024-08-30:
PR fixes this by removing the extra parameter from the docs.

closed 2024-09-08 by revarbat