BelfrySCAD/BOSL2 #1416: Sparse shapes beyond wall

jdanders, 2024-04-04:
**Is your feature request related to a problem? Please describe.**
I love the strength and efficiency of the sparse wall, but not everything I do is oriented as a wall.

**Describe the solution you'd like**
At this point, I'd like to see a sparse cuboid, but if it makes sense, this could be expanded to other 3d shapes.

**Describe alternatives you've considered**
Here's my lame version of a sparse cuboid. It assumes the holes should align with the shorted dimension. A more elegant version might have a parameter that chooses hole orientation. I would love it if all the features of cuboids were available, like rounding.

**Example Code**
```
include <BOSL2/std.scad>
include <BOSL2/walls.scad>
module sparse_cuboid(size, maxang=30, strut=5, max_bridge=20, anchor=CENTER, spin=0, orient=UP) {
  if (size[0] <= size[1] && size[0] <= size[2]) // x skinny
    attachable(anchor=anchor, spin=spin, orient=orient, size=size) {
      sparse_wall(h=size[2], l=size[1], thick=size[0], maxang=maxang, strut=strut,
                  max_bridge=max_bridge);
      children();
    }
  else if (size[1] <= size[0] && size[1] <= size[2]) // y skinny
    attachable(anchor=anchor, spin=spin, orient=orient, size=size) {
      zrot(90) sparse_wall(h=size[2], l=size[0], thick=size[1], maxang=maxang, strut=strut,
                           max_bridge=max_bridge);
      children();
    }
  else if (size[2] <= size[0] && size[2] <= size[1]) // z skinny
    attachable(anchor=anchor, spin=spin, orient=orient, size=size) {
      yrot(90) sparse_wall(h=size[0], l=size[1], thick=size[2], maxang=maxang, strut=strut,
                           max_bridge=max_bridge);
      children();
    }
  else echo("Failed sparse cuboid");
}

```

revarbat, 2024-04-04:
```openscadsparse_wall(h=30, l=20, thick=10, strut=1);
```

jdanders, 2024-04-04:
It's to use the triplet and manage the orientation of the holes (and ideally, the other cuboid parameters like rounding).

The rightmost shape in my example is the shape you reproduced, and is the shape oriented correctly for the sparse_wall. The other two shapes have suboptimal holes cut compared to my ideal results.

revarbat, 2024-04-04:
Making cylinders sparseable along their axis is doable, but I'm unclear that spheres or arbitrary 3D shapes would be practical.

jdanders, 2024-04-04:
Agreed on arbitrary shapes, that would be too broad. Prismoids, wedges, and octahedrons would be the other shapes from the libraries that could be practical.

I would imagine a sphere would be something like this, assuming someone smarter than me figured out how to size the holes correctly. And maybe instead of circular holes, you could do cutout so that the remaining lines were essentially the lat/long lines of the sphere.
[image]

revarbat, 2024-04-05:
Probably would get better results with a phi spiral a la `sphere_copies()`.
```openscad
difference() {
    sphere(d=100);
    sphere(d=80);
    sphere_copies(d=100, n=50, cone_ang=180) cyl(d=10, h=20, anchor=CTR);
}
```
[image]

revarbat, 2024-04-06:
Wireframed icosahedrons could be useful too.
[image]

adrianVmariano, 2024-04-10:
would a "sparse_wall_mask" that could be constrained take a polygon address some of these issues?  Then it could be attached with diff() to faces of any object.

adrianVmariano, 2024-05-11:
```
module sparse_cuboid(size, dir, strut=5, maxang=30, max_bridge=20,
    chamfer,
    rounding,
    edges=EDGES_ALL,
    except=[],
    except_edges,
    trimcorners=true,
    teardrop=false,
    anchor=CENTER, spin=0, orient=UP)
{
  dummy1=assert(is_vector(dir,3), "dir must be a 3-vector");
  count = len([for(d=dir) if (d!=0) d]);
  dummy2=assert(count==1 && len(dir)<=3, "dir must have exactly one non-zero component");
  attachable(anchor,spin,orient,size=size){
    intersection(){
      if (dir.x) 
         sparse_wall(size.z,size.y,size.x,strut=strut,maxang=maxang, max_bridge=max_bridge);
      else if (dir.y)
         zrot(90)
           sparse_wall(size.z,size.x,size.y,strut=strut,maxang=maxang, max_bridge=max_bridge);
      else
         yrot(90)
           sparse_wall(size.x,size.y,size.z,strut=strut,maxang=maxang, max_bridge=max_bridge);
      cuboid(size=size, chamfer=chamfer, rounding=rounding,edges=edges, except=except, except_edges=except_edges,
             trimcorners=trimcorners, teardrop=teardrop);
    }
    children();
  }    
}  

sparse_cuboid([10,20,30], strut=1, dir=RIGHT, rounding=3);
fwd(30) sparse_cuboid([30,20,10], strut=1, dir=UP, chamfer=1);
fwd(60) sparse_cuboid([20,10,30], strut=1, dir=FWD);
```

So something like the above?  Seems like you can get into trouble with roundings.  And also what you really want is for all the bracing to be rounded as well, which unlike the above, is nontrivial.

RAMilewski, 2024-05-13:
I agree with jdanders comment "but I can never figure out how to fix the orientation without major side-effects", and in my case at least, it applies in other parts of BOSL2.  (e.g. linear_sweep()). Hence our discussion recently on Gitter about having a "reset" for the reference frame.

adrianVmariano, 2024-05-13:
If you've built a model with objects that have children and change the orientation of an object it will screw up everything downstream.  But a "reset" will only help if the object in question has enough symmetry that it's anchors don't change when you change the orientation.  Otherwise you'll have trouble with positioning.

closed 2024-05-21 by revarbat