BelfrySCAD/BOSL2 #1422: Rounded  trapezoid

sonichy, 2024-04-26:
[image]

revarbat, 2024-04-26:
```openscadinclude <BOSL2/std.scad>
include <BOSL2/rounding.scad>
path = round_corners(trapezoid(w1=50, w2=70, h=50), r=10, closed=true);
polygon(path);
```

revarbat, 2024-04-26:
Actually, I forgot that trapezoid already has rounding available.
```openscad
include <BOSL2/std.scad>
trapezoid(w1=50, w2=70, h=50, rounding=10);
```

revarbat, 2024-04-26:
And if you need to specify a different rounding per corner, you can give the rounding= a list of four radii, for quadrants I, II, III, and IV, in order.  (X+Y+, X-Y+, X-Y-, X+,Y-, respectively.)

```openscad
include <BOSL2/std.scad>
trapezoid(w1=50, w2=70, h=50, rounding=[5, 10, 15, 20]);
```

[image]

closed 2024-04-27 by revarbat