BelfrySCAD/BOSL2 #1434: [BUG] Arc using width and thickness doesn't attach cleanly

MacksMind, 2024-05-22:
**Describe the bug**
An arc defined using width= and thickness= has a strait side along the X axis. Attaching to a square and using in difference() reveals an imperfection in the attachment. A similar imperfection is seen when using translation to position the arc. Both are sensitive to the arc parameters, and the imperfection appears and disappears as parameters change.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

w_list = [117, 120, 125, 130, 145, 150, 185];
h1 = 80;
h2 = 20;
z = 10;
r = 5;

module toast(w, join=false) {
  square([w,h1])
  if(join)
    attach(BACK, FRONT) arc(n=30, width=w, thickness=h2);
  else
    back(h1/2) arc(n=30, width=w, thickness=h2);
}

for(i = [0: len(w_list)-1]) {
  right(i*200) {
    fwd(h2 + 10) {
      linear_extrude(height = z)
      difference() {
        offset(r=r) toast(w_list[i], true);
        toast(w_list[i], true);
      }
    }
    
    back(h1 + 10) {
      linear_extrude(height = z)
      difference() {
        offset(r=r) toast(w_list[i], false);
        toast(w_list[i], false);
      }
    }
  }
};
```

**Expected behavior**
A clean outline with no internal artifacts.

**Screenshots**
[image]

**Additional context**
Outlines with internal artifacts produce GCAL errors.
- OpenSCAD Version: 2024.05.20 (openscad/openscad@8a890a9e1)
- Other libraries used: BOSL2 (f9e6e67)

adrianVmariano, 2024-05-22:
This appears to happen only with the dev version of OpenSCAD.  It also appears that it's just OpenSCAD doing it's thing where you've got two shapes that exactly share a boundary and it gets confused.  I haven't seen that happen in 2D before, but I also don't tend construct geometry that way.  If you use `attach()` and give `$overlap=0.01` then the extra junk goes away.  

It's interesting that this problem doesn't occur in the stable version.  I guess you could post an OpenSCAD issue on the OpenSCAD github and see what they say.

adrianVmariano, 2024-05-23:
I poked around a bit more and realized that sometimes the arc has an endpoint like [x,1e-15].  It appears that this is the cause of the issue.  The stable OpenSCAD must be handling such points differently than the dev version.  This sensitivity seems a bit unfortunate.  But in this case I have a fix where I ensure that arc endpoints fall exactly where they were specified.

closed 2024-05-23 by revarbat

MacksMind, 2024-05-24:
@adrianVmariano Thanks!