BelfrySCAD/BOSL2 #1436: [BUG] Wrong alignment on attach(inside=true) with set align

Gultak, 2024-05-26:
**Describe the bug**
When using `inside=true` together with `align=*` on `attach()` the object is positioned unexpectedly.

**Code To Reproduce Bug**
```
cube([50,10,50]) {
  attach(FRONT,BOTTOM,align=BOTTOM+LEFT,inset=5)
    recolor("red") cube(10);
  attach(FRONT,BOTTOM,align=BOTTOM+RIGHT,inset=5,
         inside=true)
    recolor("blue") cube(10);
}
```
For me it looks like a mismatch between the rotational anchor and the positioning (i.e. it looks like the anchor for positioning is swapped in all directions, not only the "attached" one).

For reference, `align()` handles the same situation as I'd expect.

**Expected behavior**
The blue cube should be positioned inside of the yellow cube on the bottom right side.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: none

Gultak, 2024-05-26:
Looks like it only happens, when `from=FRONT`:
```
diff()
cube(60)
  attach([FRONT,BACK,RIGHT,LEFT],BOTTOM,align=BOTTOM,
         inside=true,overlap=0.1,inset=-0.1)
    #cube(20)
      attach(TOP,TOP,overlap=0.01) cube(10);
```

adrianVmariano, 2024-05-29:
This bug is that the spin of the child is incorrect in some situations.  I believe (hope) it is fixed by the PR.

closed 2024-06-01 by revarbat