BelfrySCAD/BOSL2 #1452: clearance hole tolerance problem

adrianVmariano, 2024-07-16:
Hi folks, I'm trying to make a screw hole and a bit confused at the documentation vs the behaviour of the library

I'm doing (based on https://github.com/BelfrySCAD/BOSL2/wiki/screws.scad#module-screw_hole ):

```screw_hole("M10x2,40",thread=true, anchor=BOTTOM, tolerance="H14");```

and getting

ERROR: Assertion 'internalok' failed: "Invalid internal thread tolerance, H14.  Must have form <digit><letter>" in file BOSL2/screws.scad, line 1277

I think the docs for screw_hole are rather dense, and maybe even wrong. "H14" is a spec for a clearance hole, and that fails for me too. In the description it just says "The tolerance should be a nut tolerance for a threaded hole or a clearance hole tolerance for clearance holes." ...and then goes on to describe how to specify only clearance holes. (That don't appear to work). Down in the Arguments list in the Tolerance block, there is a link to https://github.com/BelfrySCAD/BOSL2/wiki/screws.scad#subsection-tolerance which describes the nut tolerance settings which are of the form "<digit><letter>" which the code seems to be asking for. Alas... I can't make those work either. I'm looking forward to seeing what @revarbat has to say about this.

revarbat, 2024-07-16:
Honestly, I don't know the first thing about the screw/screwhole tolerances code.

adrianVmariano, 2024-07-16:
Yeah, that's code I wrote.

adrianVmariano, 2024-07-18:
Bug with H12, H13 and H14 tolerances.  Other tolerances work.  However, code makes a **threaded** screw which requires an internal thread tolerance like 6H

closed 2024-07-20 by revarbat