BelfrySCAD/BOSL2 #1453: how to give thickness to a bezier patch?

dan-p3rry, 2024-07-17:
I'm modeling a sail, using bezier code in BOSL2.  I can create the shape I want with zero thickness, but I haven't found a way to give the shape the required thickness.  

`include <BOSL2/std.scad>
include <BOSL2/beziers.scad>
    sail_width = 80;
    sail_length = 0.7*sail_width;
    center_z = 20;
    edge_z = 8;

    patch = [
        [[0, 0.5*sail_width, 0], [0.33*sail_length,  0.36*sail_width, edge_z], 
                [0.67*sail_length, 0.15*sail_width, edge_z], [sail_length, 0, 0]],
        [[0, 0, 0], [0.33*sail_length, 0, center_z], 
                [0.67*sail_length, 0, center_z-5], [sail_length, 0, 0]],
        [[0, -0.5*sail_width, 0], [0.33*sail_length, -0.36*sail_width, edge_z], 
                [0.67*sail_length, -0.15*sail_width, edge_z], [sail_length, 0, 0]]
    ];
    //debug_bezier_patches(patches=[patch], size=1, showcps=false);
`

The following code is my inelegant attempt and barely satisfactory (because I'm not getting a uniform thickness), so I'm asking for help.  Thanks!
`    vnf = bezier_vnf([
        patch,
        up(1, patch),
    ]);
                difference() {
                    hull() vnf_polyhedron(vnf);
                    translate([0, 0, -0.75]) scale([1.01, 1.01, 1.00]) hull() vnf_polyhedron(vnf);
                }
`

adrianVmariano, 2024-07-19:
```
function bezier_sheet(patch, thickness, splinesteps=16, style="default") =
  assert(is_bezier_patch(patch))
  assert(all_nonzero([thickness]), "thickness must be nonzero")
  let(
        splinesteps = force_list(splinesteps,2),
        uvals = lerpn(0,1,splinesteps.x+1),
        vvals = lerpn(1,0,splinesteps.y+1),
        pts = bezier_patch_points(patch, uvals, vvals),
        normals = bezier_patch_normals(patch, uvals, vvals),
        dummy=assert(is_matrix(flatten(normals)),"Bezier patch has degenerate normals"),
        offset = pts + thickness*normals,
        allpoints = [for(i=idx(pts)) concat(pts[i], reverse(offset[i]))],
        vnf = vnf_vertex_array(allpoints, col_wrap=true, caps=true, style=style)        
  )
  thickness<0 ? vnf_reverse_faces(vnf) : vnf;
```

Try the above.  I'm not sure what to call it.  Any suggestions?

dan-p3rry, 2024-07-22:
Thanks!  I had to do some debugging and repair of my patch, then it worked perfectly.  The bezier documentation is very good, too, helped me to debug the issue on my own.  
My suggestion for the function name is bezier_patch_extrude().

adrianVmariano, 2024-07-22:
My concern about using "extrude" or "sweep" in the name is that this function can only make "thin" or "short" extrusions.  If you set the thickness to a large value and there is any negative curvature, the result will be invalid.

closed 2024-07-27 by revarbat