BelfrySCAD/BOSL2 #1456: [BUG] Parser Error on structs.scad when trying to run: `struct_set`

stiggy87, 2024-07-20:
**Describe the bug**
When trying to include the `screws.scad` and create a `screw()` or a standard `struct` get the following error:

```
Parsing design (AST generation)...
[ERROR: Parser error: syntax error in file structs.scad, line 48](.../OpenSCAD/libraries/BOSL2/structs.scad)
Execution aborted
```

**Code To Reproduce Bug**
```
#include <BOSL2/std.scad>
#include <BOSL2/structs.scad>

$fn=32;

str = struct_set(struct, "answer", 42);
```

**Expected behavior**
It should not be throwing an error.

**Screenshots**
N/A

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: [OpenSCAD](https://www.openscad.org/) version 2021.01
- Other libraries used: None.

adrianVmariano, 2024-07-20:
The parse error is that you put in those '#' characters.  OpenSCAD isn't C.  

Without then you get the error that "struct" is unknown, which is what should happen.

stiggy87, 2024-07-21:
Thanks! It worked with just one include, so not sure why it wasn't throwing the error earlier. Closing as not a bug.

closed 2024-07-21 by stiggy87