BelfrySCAD/BOSL2 #1466: worm face gears

keredson, 2024-08-19:
would be very useful!

example:
https://ntrs.nasa.gov/api/citations/20020092092/downloads/20020092092.pdf

[image]

adrianVmariano, 2024-08-20:
What is the benefit of this compared to a hypoid bevel gear?   Basically, why are these specifically useful?  

This looks interesting and if there's a good case for it then it could happen eventually, but at the moment, basic worm gears are broken and we don't know how to fix them, so this isn't going to happen soon.   I skimmed over the attached document and it is very technical.  I couldn't tell if it clearly lays out how to construct the form for these gears.

keredson, 2024-08-23:
> What is the benefit of this compared to a hypoid bevel gear? Basically, why are these specifically useful?

just higher gear ratios AFAICT.  bummer on worm gears being broken.

adrianVmariano, 2026-11-09:
Worm face gear is implemented.

closed 2026-11-09 by adrianVmariano