BelfrySCAD/BOSL2 #1468: cyl() parameter to allow shifting of anchors to align with faces

AndyLevesque, 2024-08-27:
**Is your feature request related to a problem? Please describe.**
When generating a octo prism with cyl(5,10, $fn=8), the anchors appear on the edges. A user may want the anchors on the faces.

**Describe the solution you'd like**
Add a parameter that accepts an angle and uses that angle to rotate the anchors around the cyl() to meet the users needs.

rotate (22.5) cylinder(h=h, r1=r, r2=r, $fn=8, anchor=anchor) rotate (-22.5);

**Describe alternatives you've considered**
Alternatively, since this is most likely to apply to prisms with side counts divisible by 4 and there will be few use cases beyond octoprism, create a dedicated 3D shape for octoprism that leverages cyl(); 

**Example Code**
module oct_prism(h, r=0, r1=0, r2=0, anchor=CENTER) {
    if (r != 0) {
        // If r is provided, create a regular octagonal prism with radius r
        rotate (22.5) cylinder(h=h, r1=r, r2=r, $fn=8, anchor=anchor) rotate (-22.5) children();
    } else if (r1 != 0 && r2 != 0) {
        // If r1 and r2 are provided, create an octagonal prism with different top and bottom radii
        rotate (22.5) cylinder(h=h, r1=r1, r2=r2, $fn=8, anchor=anchor) rotate (-22.5)  children();
    } else {
        echo("Error: You must provide either r or both r1 and r2.");
    }
}

**Additional context**
Add any other context or screenshots about the feature request here.

adrianVmariano, 2024-08-27:
The anchors aren't "on the edges".  They are on the ideal cylinder.  This definitely seems like not what you want for small $fn values.  

I'm not sure what the solution to this is.  Should we make anchors stick to the true object for all $fn?    I do think that the natural way to get anchors on the faces for $fn divisible by 4 would be to use cyl() instead of cylinder() and use realign=true.  If anchors then appeared on the object (instead of on the cylinder) you'd get what you want.  

Hmmm.  This seems a bit connected to another interface problem, the fact that textures force $fn.  Maybe the answer to both problems really is to create a prism() module, where cyl() stays as it is with anchors on the ideal object, and with textures sampled under the assumption that you're making a round object, and the prism() module gives anchors on the object and the specified number of faces, with texturing on planar faces.  Then we could also supply anchors for every edge and face in the prism case the same way that regular_ngon() does.  

I'm feeling more and more like adding prism() or something like that is the right answer.  @revarbat What do you think?

AndyLevesque, 2024-08-28:
> The anchors aren't "on the edges". They are on the ideal cylinder. This definitely seems like not what you want for small $fn values.
>
> I'm not sure what the solution to this is. Should we make [...]

You make a strong case for not modifying cyl() and instead having a separate prism() module. That would address my concern and be a powerful addition.

adrianVmariano, 2024-08-28:
This idea also addresses another problem of rounding the ends of a prism.  If you set $fn=4 with cyl() and use rounding you only get $fn=4 for your end rounding.  

This does raise a question though about how edge and face anchors should work with rounding.  Do we have two modes, one where anchors are on the ideal prism and one where anchors are on the actual object?   Or two different sets of anchors for the faces and edges?  (Not sure what they'd be called.)  And is the same thing needed for the basic anchors, which I thought could inherit just from VNF anchors?

adrianVmariano, 2024-08-30:
This has been added in the PR.  Let me know if you find any issues with the new version once it is merged and you are able to try it out.  I created named anchors for sides, edges and corners of the prism, similarly to named anchors for regular_ngon().

closed 2024-09-08 by revarbat