BelfrySCAD/BOSL2 #1470: [BUG] planetary_gears() Documentation error

dan-p3rry, 2024-09-10:
**Describe the bug**
The description of sun_ring and ring_sun should be swapped as far as which one is >1 and <1.  (The documentation sounds correct, but I have to set sun_ring=6 to get ring gear 6x the sun gear.)  
Also in this section, not exactly an error, the variables ring_carrier, sun_carrier, etc are inconsistent with the text, where the term "planet" is used. 

Thanks a lot for these gear functions.  I was really beating my head trying to figure this stuff out.  Turns out I was close, but not converging.

adrianVmariano, 2024-09-10:
Gears are remarkably complex and subtle.  Are you saying that when you set sun_ring=6 you actually get a sun/ring ratio of 1/6?  Because that sounds like an implementation error.

dan-p3rry, 2024-09-11:
No, I am saying that sun_ring=6 works, but the documentation says:  set
sun/ring ratio to this value in a sun driven system, *must have absolute
value smaller than 1*

That bold text should say "must have absolute value greater than 1".
 Setting the value to less than one causes an error in the console.

(I would argue that the arguments sun_ring and ring_sun are named
incorrectly (should be swapped), because the sun is always smaller than the
ring, but fixing that would break someone's code if they updated BOSL2.)

Dan

dan-p3rry, 2024-09-11:
Re-reading your response, yes you are correct that setting sun_ring = 6
actually generates sun/ring ratio of 1/6.
Dan

adrianVmariano, 2024-09-11:
That is a bug then and will need to be fixed.

adrianVmariano, 2024-09-11:
I investigated more and it appears to me that setting sun_ring=6 produces the specified transmission ratio that the sun gear rotates six times for every single rotation of the ring gear.

dan-p3rry, 2024-09-12:
Ok, so we are back to just the documentation error then.

adrianVmariano, 2024-09-13:
Yeah, I have corrected the docs and also clarified that "ratio" means "transmission ratio".

dan-p3rry, 2024-09-15:
Thanks for the fix

closed 2024-09-15 by dan-p3rry