BelfrySCAD/BOSL2 #1471: [BUG] offset is path direction dependent

pkriens, 2024-09-12:
**Describe the bug**
When using offset as function the operation depends on if the path is moving left or right. If the path moves left, a positive delta or r is decreasing the size of the path. If the path moves right, it works as expected. This behavior is different from the builtin offset.

Making this example, I also noticed the shape was deformed in the function. The figure is symmetric before `offset` but it isn't after. The angled side top and bottom have different size.

**Code To Reproduce Bug**
```
            path = [
                [5, -2], 
                [6, -1], 
                [6, 1], 
                [5, 2], 
                [-5, 2], 
                [-6, 1], 
                [-6, -1], 
                [-5, -2]
                ];
            delta = -1;
            %linear_extrude(0.2) polygon(path);
            linear_extrude(0.1) offset(delta=delta) polygon(path);
            fwd(8) {
                %linear_extrude(0.2) polygon(path);
                linear_extrude(0.1)  polygon(offset(delta=delta,path));
            }
```
**Expected behavior**
I expect the behavior to be identical to the builtin `offset` function with the same parameters.

**Screenshots**
[image]

When the path is reversed 

[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 
[image]

BOSL2 date is Aug 7 2024

- Other libraries used: none

adrianVmariano, 2024-09-12:
For open paths this is working as intended.  There is no sensible way to make it direction independent.   Note that results will be possibly be wrong if you repeat the point the way you have done because the normal won't take into account what happens at the other end of the path.    You need to use `closed=true`.   (Arguably the default value for `closed` is wrong and it should match the builtin.)

pkriens, 2024-09-13:
Ah, that explains it. I missed the closed.  Might want to highlight that in the documentation?

closed 2024-09-13 by pkriens