BelfrySCAD/BOSL2 #1473: Trapezohedron feature request

damonvp, 2024-09-19:
It would be great if it was possible to choose the alignment of the upper and lower sections.
The problem with some trapezohedrons is that if used for dice the flat face down opposes the middle of 2 opposing faces instead of an opposing face, thereby there is no number to read.
Specifically making it useless for a d16.
I guess that would be an octagonal bipyramid?

adrianVmariano, 2024-09-19:
Are you requesting a shape with triangular faces?  That wouldn't be a trapezohedron, which is a shape with four sided faces.  If I understand what you're after, you can make the shape you want using to uses of cyl() or cylinder(), or one with a flip_copy().

damonvp, 2024-09-20:
thanks I think this will work:
n = [0,0,1];
 module obj()regular_prism(8, side1=10, ir2=0, height=20, anchor=BOTTOM);
up(20) obj();
up(20) mirror(n)obj();

damonvp, 2024-09-20:
Oh, I lost the ability to use draw=false for the numbers :(

adrianVmariano, 2024-09-20:
You should be able to anchor to the prism faces instead.

adrianVmariano, 2024-09-20:
Unless you needed numbers with no object?

damonvp, 2024-09-23:
Admittely, Im a noob here... Im trying to update PolydiceGenerator and they is the method used to make the numbers.
Ive tried doing numbers other ways and gotten stumped pretty quickly.
Here is the code they use:
module drawd16(){
    txt_merged=merge_txt(d16_text,fix_quotes(d16_symbols));
    txt_mult=d16_text_size*d16_size/200;
    adj_txt=adj_list(d16_adj_size,d16_size/100);
    txt_stroke=text_stroke*txt_mult;
    sym_mult=d16_symbol_size*d16_size/100;
    sym_stroke=symbol_stroke*sym_mult;
    bumpers=fix_quotes(d16_bumpers);
    under_mult=d16_underscore_size*d16_size/100;
    space_mult=d16_text_spacing>1 ? (d16_text_spacing-1)*txt_mult/3.15 : d16_text_spacing<1 ? (-1+d16_text_spacing)*txt_mult/2.8 : 0;
    base_rotate=[16.813,16.813,253.682,343.682,253.682,343.682,253.682,254.289,106.318,254.289,16.318,254.289,343.682,106.289,343.682,343.682];

    d16_height = sqrt(0.5 * pow(d16_size, 2)) -d16_length_mod/d16_size;
    circumsphere_dia=regular_polyhedron_info("out_radius", "trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM) * 2;
    corner_round_mult=circumsphere_dia-(corner_rounding*circumsphere_dia/100)/1.6;
    corner_clip_mult=d16_size-(corner_clipping*d16_size/100)/1.2;
    
    difference()
    {
        if(add_bumpers && edge_rounding==0 && corner_rounding==0 && corner_clipping==0)
            //render bumpers
            union()
            {
                regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM);
                regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM,rotate_children=false,draw=false)
                if(bumpers[$faceindex]) stroke($face,width=bumper_size,closed=true);
            }
        else if(edge_rounding==0 && (corner_rounding>0 || corner_clipping>0))
            //render clipping objects
            intersection()
            {
                regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM);
                if(corner_rounding>0)
                    translate([0,0,d16_size/2])
                    sphere(d=corner_round_mult);
                else if(corner_clipping>0)
                    translate([0,0,d16_size/2])
                    rotate([0,-48,36]) rotate([0,0,54])
                    pentagonal_antiprism(corner_clip_mult);
            }
        else
            //render trapezohedron 
            regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM,rounding=edge_rounding);

        //render numbers & symbols
        regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM,draw=false)
        zrot(d16_rotate[$faceindex]+base_rotate[$faceindex])
        down(text_depth+d16_adj_depth[$faceindex])
        linear_extrude(height=2*text_depth+d16_adj_depth[$faceindex])
        move([(d16_text_h_push+d16_adj_h_push[$faceindex])*d16_size/100,(d16_text_v_push+d16_adj_v_push[$faceindex])*d16_size/100])
        if(is_list(txt_merged[$faceindex])) //a symbol
            move([d16_symbol_h_push*d16_size/100,d16_symbol_v_push*d16_size/100])
            offset(delta=sym_stroke)
            text(txt_merged[$faceindex][0],size=sym_mult,font=sym_font,halign="center",valign="center");
        else if(txt_merged[$faceindex]=="4") //a number 4
            right(d16_num_4_h_push*d16_size/100)
            offset(delta=txt_stroke)
            text(txt_merged[$faceindex],size=txt_mult+adj_txt[$faceindex],font=txt_font,halign="center",valign="center");
        else if(len(txt_merged[$faceindex])==1) //a single digit number that's not 4
            offset(delta=txt_stroke)
            text(txt_merged[$faceindex],size=txt_mult+adj_txt[$faceindex],font=txt_font,halign="center",valign="center");
        else //a double digit number
            right(space_mult)
            offset(delta=txt_stroke)
            text(txt_merged[$faceindex],size=txt_mult+adj_txt[$faceindex],font=txt_font,spacing=d16_text_spacing,halign="center",valign="center");
        
        //render underscore
        regular_polyhedron("trapezohedron",faces=16,h=d16_height,longside=d16_size,anchor=BOTTOM,draw=false)
        zrot(d16_rotate[$faceindex]+base_rotate[$faceindex])
        down(text_depth+d16_adj_depth[$faceindex])
        linear_extrude(height=2*text_depth+d16_adj_depth[$faceindex])
        move([d16_underscore_h_push*d16_size/100,d16_underscore_v_push*d16_size/100])
        offset(delta=txt_stroke)
        text(d16_underscores[$faceindex],size=under_mult,font=under_font,halign="center",valign="center");
    }
}

adrianVmariano, 2024-09-23:
That code is missing stuff and doesn't run as-is.  It's not great.   The only reason you ever need draw=false, as far as I can think, is that the actually polyhedron DOES NOT APPEAR in your model.  Otherwise, you should create the polyhedron and create all the marks and numbers and bumpers and whatnot as children to the polyhedron.  If you modify the code to do that then from there you can jump to making a pyramid and using the face### anchors instead.  It's a little clumsier in that you'll need something like:

```
regular_prism(n=8, h=10, side1=5, side2=0.5)
  for(i=[0:7])
    attach(str("face",i)) text3d(str(i), thickness=1,size=2,anchor=CENTER);
```

closed 2024-09-23 by adrianVmariano

damonvp, 2024-09-23:
thanks, but how do I use this method to carve out the numbers, not have them sit on top?

adrianVmariano, 2024-09-23:
Consider reading the attachments tutorial.  

```
diff()
regular_prism(n=8, h=10, side1=5, side2=0.5)
  for(i=[0:7])
    tag("remove")attach(str("face",i)) text3d(str(i), thickness=1,size=2,anchor=CENTER);
```