BelfrySCAD/BOSL2 #1474: shaped2d text module defaults to Helvetica, which is not universally available

MethylBromide, 2024-09-22:
**Describe the bug**
If a call to the **text** module of shapes2d.scad is made without providing a **font** value, the default font used is Helvetica, which isn't available on all systems and which produces odd results.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
text("abcd");
```

**Expected behavior**
I had expected the same default font as the built-in text function to be used, or at any rate some reasonable, legible, sans-serif font. Instead an apparently random font is used since nothing on my system is called Helvetica. The font used is "Goma Stencil" so the name doesn't particularly resemble "Helvetica".

I don't understand why the text module in BOSL2 needs to override the default font of the **built-in** text function. It's a perfectly fine font.

Now there is _also_ an OpenSCAD bug here because replacing an unrecognized font with a random one isn't reasonable behavior. But I also think BOSL2 is wrong to specify Helvetica -- especially since if you hope to remain backward compatible to 2019 versions, you should work around this bug.

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2024.09.17 (git 399551768)
- Other libraries used: none
- OS: Windows 11 64bit on Intel PC.

revarbat, 2024-09-22:
... in days of Yor it used to be that there were exactly three universal platform independent fonts: Helvetica for sans-serif, Times for serif, and Courier for monospaced. In TCL/Tk, in fact, those font names were internally remapped to the platform similar font, if it somehow DIDN'T exist. My last big project was a CAD/CAM written in TCL/Tk, so I apparently made that assumption when I wrote that module. My bad. Perhaps if we default to passing `font=undef`, we can get the platform default font?

adrianVmariano, 2024-09-22:
This change should presumably be made to all 3 text generating modules.  Passing font=undef does appear to work.  (Produces identical behavior on my machine to the current "Helvetica" default.)   I think the docs for the font option for those 3 modules gives incorrect information, though I guess we don't know what the default font is?   I checked with Jordan and he says OpenSCAD shipes with Liberation Sans and that should be the default font.  

It does seem like OP has something wrong with their font setup that it returns Goma Stencil when Helvetica is requested.  I suspect that's not an OpenSCAD problem.   But there's no reason for us to specify a default font like we are currently doing.

adrianVmariano, 2024-09-22:
Actually if @MethylBromide could enable the experimental textmetrics feature and run 

`echo(fontmetrics());`

that would tell us their default font and confirm that it is Liberation Sans as expected.

MethylBromide, 2024-09-22:
ECHO: { nominal = { ascent = 12.5733; descent = -2.9433; }; max = { ascent = 13.6109; descent = -4.2114; }; interline = 15.9709; font = { family = "Liberation Sans"; style = "Regular"; }; }

adrianVmariano, 2024-09-22:
Yeah, that definitely sounds like your system is broken with regards to font management.   Pretty strange.  I've got the default font fixed in a PR.  

Can you check if "Liberation Mono" works for you?  I'm told the liberation fonts will always work, so I'm thinking of changing all the "Courier" examples in the manual to "Liberation Mono" instead to ensure that examples work for everybody.

MethylBromide, 2024-09-23:
"Liberation Sans" is used when I specify it explicitly or when I use the system text function.
"Liberation Mono" works as expected.
I don't understand how the fontmetrics output leads you to think something is special about my system.

adrianVmariano, 2024-09-23:
What seems special about your system is that it returns "Goma Stencil" when any unavailable font is requested.  The font system is supposed to select a font with the same characteristics as the requested font, not return a weird special effects font.  It should work to request Helvetica on your system.  That it fails indicates that something is broken somewhere.   

The font metrics output just gives me confirmation that things are working as expected and that if I claim that "Liberation Sans" is the default font it's not false somewhere.

revarbat, 2024-09-23:
Fix has been merged into mainline.

closed 2024-09-23 by revarbat