BelfrySCAD/BOSL2 #1479: [BUG] Bug in drawing.scad function arc

mrlocke, 2024-09-28:
**Describe the bug**
There is a bug in a drawings.scad function arc, when called with corner points in 3d.

**Code To Reproduce Bug**
```
corner=[[-13.6667, 0, 15.5], [-13.6667, 0, 40.5], [-31.3443, 0, 58.1777]]; // this really does not matter, as long as it is using 3d
tl1tl2 = arc(40, r=15, corner=corner); // only corner
```

**Expected behavior**
A path should be returned. Insted it thowns an error:
```
[ERROR: Assertion '(is_finite(r) && (r > 0))' failed: "Must specify r= or d= when corner= is given." in file ../../../../.local/share/OpenSCAD/libraries/BOSL2/drawing.scad, line 795](https://github.com/BelfrySCAD/BOSL2/issues/795,/home/locke/.local/share/OpenSCAD/libraries/BOSL2/drawing.scad)
```

Reviewing the code, I've found that if 3d corner detected, it is converted into 2d, and a function is recursively called with 2d parameters on line 790:
```
lift_plane(plane,arc(n,corner=points2d,wedge=wedge,long=long))
```
Without `r` or `d` parameters, but instead with not allowed in this call `wedge` and `long`. Replacing this 2 parameters solved the issue.

Working line:
```
lift_plane(plane,arc(n,corner=points2d,r=r,d=d))
```

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: version 2021.01 
- BOSL2 version: BOSL_VERSION = [2,0,716];
- Other libraries used: include <BOSL2/std.scad>

adrianVmariano, 2024-09-28:
Actually `wedge` is allowed but not `long`.  I've got the fix in my PR.

mrlocke, 2024-09-28:
Thank you! Really appreciated! :-)

closed 2024-10-05 by adrianVmariano