BelfrySCAD/BOSL2 #1493: [BUG] empty space is created between threads and shank when using thread_len

TheRealBenForce, 2024-10-19:
**Describe the bug**
I'm attempting to make a screw similar to the one on the right side of this picture:
https://github.com/BelfrySCAD/BOSL2/wiki/images/screws/figure_2_1_1.png

However there is negative space between the shank and threads.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
screw("M6", head="flat", drive="hex", length=20, thread_len=10, orient=DOWN);
```

**Expected behavior**
One solid object is created with a thread length half the size of the screw length

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: none

adrianVmariano, 2024-10-19:
I tried your sample code and got this result:

[image]

which appears to be OK.  Are you using a recent BOSL2?

TheRealBenForce, 2024-10-21:
TL;DR - Looks like an include of a file that has an include of BOSL (not BOSL2) is causing a conflict. 
**Are BOSL and BOSL2 known to cause issues in the same project?**

@adrianVmariano , apologies on the delay. Had some out of town guests staying in my office.

Yes I'm using what is in master branch of BOSL2. So my code sample I pasted had some what I thought were unnecessary bits stripped out which I now discovered are what is actually causing the issue. Please look at the 3rd line in screws.scad

screws.scad
```scad
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
include <arm.scad> // renders fine when this line is commented out.
include <globals.scad>

$fn=32;

module hex_screw() {
    color("gray")
    screw("M6", head="flat", drive="hex", length=20, thread_len=10, orient=DOWN);
}

hex_screw();
```

arm.scad
```scad
include <BOSL/shapes.scad>
// Created this file using BOSL before I began using BOSL2. 
...
```

I get these 3 errors when I'm importing arm.scad. Without the import, no errors and it renders fine.
[image]

adrianVmariano, 2024-10-21:
I would absolutely expect problems if you try to combine BOSL and BOSL2 in one project, but I have not tried to characterize the nature of the problems.  Perhaps you can use<BOSL/shapes.scad> instead of including?  I don't know if that will work or not, but that would be the only possible way it could work.   The errors indicate to me that BOSL has overridden shapes2d.scad and screws.scad, so anything that depends on those files in BOSL2 may malfunction.

TheRealBenForce, 2024-10-22:
Cool, well thanks for validating my initial code! I'll close this out.

closed 2024-10-22 by TheRealBenForce