BelfrySCAD/BOSL2 #1496: [BUG] align/attach are not working if the parent is a module

pallaire, 2024-10-29:
**Describe the bug**

When I try to use attach or align from BOSL2 to create and align children parts, they are not created if the parent is a module. It works fine if the parent is a primitive.

In the screenshot below: 
The red part, is parent=primitive & children=primitive ... works fine
The purple part, is parent=primitive & children=module ... works fine
The cyan part, is parent=module & children=module ... Doesn't work
The red part, is parent=module & children=primitive ... Doesn't work 

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

ext15 = 1.5;
length = 24;

module Ext1515x(length) {
    cube([length,1.5,1.5], anchor=CENTER);
}

module Ext1515y(length) {
    cube([1.5,length,1.5], anchor=CENTER);
}

module Ext1515z(length) {
    cube([1.5,1.5,length], anchor=CENTER);
}

back(5)
color("red")
cube([length, ext15, ext15], anchor=CENTER) {
  attach(TOP, BOTTOM, align=[LEFT])  cube([ext15, ext15, length], anchor=CENTER);
  attach(TOP, BOTTOM, align=[RIGHT]) cube([ext15, ext15, length], anchor=CENTER);
}

back(10)
color("purple")
cube([length, ext15, ext15], anchor=CENTER) {
  attach(TOP, BOTTOM, align=[LEFT])  Ext1515z(length);
  attach(TOP, BOTTOM, align=[RIGHT]) Ext1515z(length);
}

color("cyan")
Ext1515x(length) {
  attach(TOP, BOTTOM, align=[LEFT])  Ext1515z(length);
  attach(TOP, BOTTOM, align=[RIGHT]) Ext1515z(length);
}

fwd(5)
color("yellow")
Ext1515x(length) {
  attach(TOP, BOTTOM, align=[LEFT])  cube([ext15, ext15, length], anchor=CENTER);
  attach(TOP, BOTTOM, align=[RIGHT]) cube([ext15, ext15, length], anchor=CENTER);
}
```

**Expected behavior**
Align, attach, position ... should all create the children if the parent is a module as if the parent was a primitive.

**Screenshots**
[image]

**Additional context**
- OpenSCAD Version: 2024.10.16
- BOSL_VERSION = [2,0,716];

adrianVmariano, 2024-10-29:
Attachment only works with attachable objects.  You did not make your modules attachable by calling the attachable() module in them when creating the geometry, so of course it will not work.   In this case because your modules are just passthroughs for attachable modules you can get away with just invoking the children.  (If you thought this could possibly work without somehow invoking the children in your modules you might want to read up on OpenSCAD basics.  You passed children into modules and then never invoked them, so naturally they just vanish.)  

So something like:

```
module foo(length) {
    cube([1.5,length,1.5], anchor=CENTER)
      children();
}
```

will work without a full call to attachable()

closed 2024-10-29 by adrianVmariano

pallaire, 2024-10-29:
Nice, got it, it works. I did read the children() doc before, while searching for a solution to my problem. I don't know why, but I didn't think it would apply here. I was mostly looking at a parent() call for the child, rather than a children() call for the parent. 🤷‍♂️