BelfrySCAD/BOSL2 #1508: [BUG] regular_polyhedron_info "index set" query throws inappropriate error

MethylBromide, 2024-11-19:
**Describe the bug**
A call to **regular_polyhedron_info** requesting the "index set" for a value of **faces** for which there are no known regular polyhedra, should return an empty set -- not throw an Assert error.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/polyhedra.scad>
echo(regular_polyhedron_info(faces=21, info="index set"));
```

**Expected behavior**
Expected something like:
`ECHO: []`
to show that the result is the empty set. Instead I get:
```
Compiling design (CSG Tree generation)...
[ERROR: Assertion '(len(indexlist) > 0)' failed: "No polyhedra meet your specification" in file ../Program Files/OpenSCAD/libraries/BOSL2/polyhedra.scad, line 660](https://github.com/BelfrySCAD/BOSL2/issues/660,C:/Program%20Files/OpenSCAD/libraries/BOSL2/polyhedra.scad)
[TRACE: called by 'regular_polyhedron_info' in file , line 4](https://github.com/BelfrySCAD/BOSL2/issues/4,C:/3d)
[TRACE: called by 'echo' in file , line 4](https://github.com/BelfrySCAD/BOSL2/issues/4,C:/3d)
```
The file polyhedra.scad already contains all the knowledge about the properties of its supported polyhedra. It doesn't make sense to duplicate this information in the calling code to know which values of **faces** will cause emergency shutdown. This is not an actual emergency.

I mean, damn, I was just asking a question, you don't have to shoot me for it. If there are no matching polyhedra, I would like for _my code_ to be able to decide how to recover from that gracefully -- not to have it decided for me. The abort is not appropriate.

Sure, I could put something in my code containing a list of the valid values that will not cause an error, and I guess that's what I'll have to do until this is fixed, but that's a maintainability problem. What if they add some new shapes to the library and now the library is no longer in sync with the calling code?

- OpenSCAD Version: 2024.10.28 (git 7f05c6294)
- Other libraries used: none

adrianVmariano, 2024-11-20:
Seems reasonable.

adrianVmariano, 2024-11-21:
This has been fixed by the latest PR.

closed 2024-11-21 by adrianVmariano