BelfrySCAD/BOSL2 #1510: Best way to round everything

davidedelvento, 2024-11-20:
Consider the following:

```c
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

hole = 1.3;
width = 9;
thickness = 5;
beam_len = 13;
mylen = 20;

difference() {
  union() {
    up(beam_len/2 + mylen/2)
    rounded_prism(rect([width, thickness]), height=mylen, joint_top=thickness/8,
                                                          joint_bot=thickness/8,
                                                          joint_sides=thickness/8,
                                                          k = 0.9,
                                                          anchor=FRONT);
    rounded_prism(rect(thickness), height=beam_len, joint_top=-thickness/8,
                                                    joint_bot=thickness/8,
                                                    joint_sides=thickness/8,
                                                    k = 1,
                                                    anchor=FRONT);
  }
  rotate([0,90,0])
  translate([beam_len/3.5,thickness/2-hole/2,0])
  cyl(l=thickness+0.01, d=hole, $fn=32, rounding=-hole/3, anchor=FRONT);
}
```
which generates 

[image]

The way the two prisms attach is not what I want (which is the "smoothest" possible object). Zooming in the transition region:

[image]

Of course I can avoid rounding the top/bottom to match as in

```c
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

hole = 1.3;
width = 9;
thickness = 5;
beam_len = 13;
mylen = 20;

difference() {
  union() {
    up(beam_len/2 + mylen/2)
    rounded_prism(rect([width, thickness]), height=mylen, joint_top=thickness/8,
                                                                 joint_bot=0,
                                                                 joint_sides=thickness/8,
                                                                 k = 0.9,
                                                                 anchor=FRONT);
    rounded_prism(rect(thickness), height=beam_len, joint_top=0,
                                                    joint_bot=thickness/8,
                                                    joint_sides=thickness/8,
                                                    k = 1,
                                                    anchor=FRONT);
  }
  rotate([0,90,0])
  translate([beam_len/3.5,thickness/2-hole/2,0])
  cyl(l=thickness+0.01, d=hole, $fn=32, rounding=-hole/3, anchor=FRONT);
}
```

which makes the top/bottom match better, but now many rounding (and a fillet) that I *do* want are missing:

[image]

The question is: how do I round every external corner and fillet every internal corner of (potentially much more) complicated objects made of various parts?

I can think of many ways:

* A "magic" `round_and_fillet_everything` function, like `rotate`, `color` and the likes. That would be fantastic, but I doubt it exists, otherwise I would have found it in the docs and I didn't
* Sapient usage of [`join_prism`](https://github.com/BelfrySCAD/BOSL2/wiki/rounding.scad#functionmodule-join_prism): that can do more complicated stuff than this, so it must be possible to use it in this context. The examples are all much more than I need here, and not easy to understand for a newbie like me. I'm eager to learn them, but only if that's the best way to proceed.
* More sophisticated use of `rounded_prism` than I am doing now. The simpler I can think of is adding another beam like the one on the bottom, slightly shorter but raised up a little more: that would take care of the groove caused by the roundness of the top prism. Then I could subtract two more non-rounded prism from each side, to remove the unnecessary fillet from those two sides. I can do that now, and I would not need to learn anything new. But it would become increasingly complex the more complex the part becomes, and it seems tedious and contrary of the simpleness of BOSL philosophy (which I appreciate a lot) -- even though [attachments](https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad) (which I should be already using rather than hard-coding the translation) could make it less tedious and clearer
* Using attachments for creating the desired [edge_profile](https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-edge_profile) or [edge_mask](https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-edge_mask) or the corresponding corner ones or [mask_ogee](https://github.com/BelfrySCAD/BOSL2/wiki/masks2d.scad#functionmodule-mask2d_ogee) or one of the various related ways

Which approach is the most natural, simpler, most maintainable, and that most generalize to arbitrary objects made of many parts? For context I don't want to do fancy shapes (e.g. like the ones that look like decorative edges created with specialty decorative routers), but I want to simply design parts for 3D manufacturing in a way that avoid the brittleness, sharpness and difficulties of creating non-rounded corners.

TIA!!

davidedelvento, 2024-11-20:
To be clear what I meant with option 3 and what I would like to achieve, here is a naive example adding/subtracting those other prisms

[image]

In fact, even so it's not perfect but there is that small piece of fillet in the corner which does not get removed

adrianVmariano, 2024-11-21:
Generally speaking there is no one universal way to round things that always solves the problems in OpenSCAD.  This example has a troublesome corner where the two prisms meet, but because it has two planar faces the whole object is actually a prism, hence in this case, the best answer is something along these lines:

```
base=union([rect(20), right(15,rect(10))]);
rounded_prism(base[0], h=15, joint_top=1, joint_bot=1, joint_sides=1);
```

which produces:

[image]

I think this approach, or the analogous one using offset_sweep are the only methods that can round that concave-convex corner.   Or I guess maybe a complex scheme involving path_sweep.

closed 2024-11-21 by adrianVmariano