BelfrySCAD/BOSL2 #1515: [BUG] Slider anchors are incorrect

Skinner927, 2024-12-04:
**Describe the bug**

1. `slider()`'s anchor box is rotated 90 degrees and too tall.
2. `slider()`'s default spin should match `rail()`.

**Code To Reproduce Bug**

```
include <BOSL2/std.scad>
include <BOSL2/sliders.scad>

slider() show_anchors();

right(40) rail() show_anchors();
```

[image]

**Expected behavior**

Anchors should lay correctly on `slider()` and it should be oriented the same as its matching `rail()`.

**Fix**

I can make a PR if you prefer.

[image]

```diff
diff --git a/BOSL2/sliders.scad b/BOSL2/sliders.scad
index be15ee6..58e2a68 100644
--- a/BOSL2/sliders.scad
+++ b/BOSL2/sliders.scad
@@ -41,9 +41,8 @@ module slider(l=30, w=10, h=10, base=10, wall=5, ang=30, anchor=BOTTOM, spin=0,
     full_width = w + 2*wall;
     full_height = h + base;
 
-    attachable(anchor,spin,orient, size=[full_width, l, h+2*base]) {
-        zrot(90)
-        down(base+h/2) {
+    attachable(anchor,spin,orient, size=[full_width, l, full_height]) {
+        down(full_height/2) {
             // Base
             cuboid([full_width, l, base-get_slop()], chamfer=2, edges=[FRONT,BACK], except_edges=BOT, anchor=BOTTOM);
```

revarbat, 2025-03-25:
The problem here is that the vertically centered anchors need to line up with the centerline of the slider V wedges.  If you provide the correct `offset=`, you can have the top and bottom anchors at the corners of the structure as shown here, and have the vertically centered anchors aligned with the tip of the slider V's.

adrianVmariano, 2025-05-04:
You didn't change the orientation as requested.  Is that because you think it's right?  Or because of backward compatibility concerns?

revarbat, 2025-05-04:
Backwards compatibility.

revarbat, 2025-05-04:
Honestly this is a rarely enough used feature that we can probably get away with fixing the spin without annoying too many people. It definitely seems wrong at the moment.

adrianVmariano, 2025-05-04:
You can add a deprecation message like we've been doing.  

I was wondering if it should actually be z aligned so that orient works on it, but I don't know how one uses this.

adrianVmariano, 2025-05-04:
I've always found it annoying that teardrop is not z aligned because that seems to make it much harder to position, but I don't know if this is the same.

revarbat, 2025-05-05:
There's no true alignment, as my printable printer uses these rails and sliders along ALL three axes.

revarbat, 2025-05-05:
Not sure how to deprecate an alignment change.

adrianVmariano, 2025-05-05:
The point about z alignment is that you designed the anchoring system around objects being z aligned.  So if I want the thing X aligned I can say orient=RIGHT and if I want it Y aligned, orient=BACK.   Now alternatively one could imagine using attach() with spin to get it pointed the right way.  I don't know what's easiest or most natural for the sliders.  For teardrops I think z alignment is most natural and the existing choice of y (?) alignment is confusing, but it may be different, because when you attach a teardrop you want it perpendicular to the parent.  

For this sort of deprecation I've been doing things like:

`dummy= $slider_msg ? echo("deprecated!  Set $slider_msg=false to hide this message."):0;`

adrianVmariano, 2025-05-13:
@revarbat I reopened this so you'll remember to fix the slider orientation.

adrianVmariano, 2025-05-21:
Is this one fully resolved now?

PR #1668 (2025-05-04): Fixed slider anchors.
Closes #1515
closed 2025-05-25 by adrianVmariano