BelfrySCAD/BOSL2 #1516: [BUG] Misc documentation issues with `path_to_bezpath()` and `smooth_path()`

amatulic, 2024-12-04:
Some minor issues with documentation:

* On the Wiki page https://github.com/BelfrySCAD/BOSL2/wiki/beziers.scad#function-path_to_bezpath the function `path_to_bezpath()` has no illustrations, but it references `smooth_path()` for illustrations, which is fine. However, `[smooth_path()](https://github.com/BelfrySCAD/BOSL2/wiki/rounding.scad#function-smooth_path)` isn't linked under the "See also" heading, and it should be.
* In the documentation for `smooth_path()`:
** There is no example of the effects of changing relsize; perhaps one image could be added.
** In example 6, the word "align" is misspelled "aline".

One question: If a raise a PR to contribute to BOSL2 (I have a PR nearly ready), it's obvious how to contribute code, but how would I contribute changes to the documentation? Just include the documentation in the PR description?

adrianVmariano, 2024-12-05:
The documentation for BOSL2 is automatically generated from the comments in the BOSL2 code, so if you change those comments it will automatically update the documentation.  So fix any typos there.  You can add examples---just follow the structure of other examples for how to do that.  The key thing is to use consistent indentation, or you'll confuse docsgen.

amatulic, 2024-12-05:
Ah, I see. That explains all the comments I see. How would graphics be changed in the documentation pages? That isn't clear. Are the figures automatically generated from the commented code examples?

What about wiki pages that don't have corresponding code? For example, https://github.com/BelfrySCAD/BOSL2/wiki/AlphaIndex has "boundary" misspelled as "boundry" for `stroke()`.

adrianVmariano, 2024-12-05:
Yes, figures are automatically generated from the commented code examples.  In addition to documenting stuff it also keeps us honest---if you see an example in the wiki it works as shown, guaranteed, or it couldn't be there.    (Or to be more precise, they serve as the test cases for things that provide the possibility of identifying errors or regressions in the code, since there's no way to do unit testing on modules.)   So if you change or extend code you should add a new example that demos the new feature which both tests that it's working as desired, shows the user how to use it, and gives us a chance to catch a problem down the road if it stops working because the example will either fail completely when the docs are regenerated, or it will produce a funny looking result that someone will hopefully notice eventually....  

The entire wiki is autogenerated by docsgen from comments and a few top level files in the source tree.

closed 2024-12-05 by amatulic