BelfrySCAD/BOSL2 #1526: Does teardrop need a bottom corner?

adrianVmariano, 2024-12-13:
There are a lot of clearance guidances published, recommending as much as 0.5 mm for a loose fit. There are going to be horizontal tolerance variations as you print, arising from:
* Variations in the filament diameter. A typical spec you see on amazon is &pm;0.03 mm, which creates variations in volumetric flow, which in turn creates variations in line width.
* Extrusion multiplier. This is filament-specific and most people don't calibrate it for each new spool they get. When I printed a cube with white silk PLA in vase mode with a line width of 0.45, I got a wall thickness of 0.5-0.54, which is considered an acceptable tolerance variation. But that means the perimeters from that filament are nearly 0.05 mm wider on each side. Generally slight overextrusion like this results in better quality parts.
* Perimeter order. Usually one prints inner perimeters first, so that outer perimeters have something on the side to stick to when printing overhangs. However, because the inner perimeter may be squished a bit wider, the outer perimeter is going to get pushed slightly outward.

Printed vertical variations from the design arise mostly from:
* Using constant increments of layer height. If your design has something 10.27 mm off the build plate, there is no way a 0.15 or 0.2 layer height can match it. I don't know the slicer algorithm, but I expect the slicer would add or omit a layer depending on which action results in the smallest error.
* First layer. Typically the first layer is 0.2 mm unless you consciously set it to something else. The Prusa default printer profiles all have 0.2 mm for the first layer regardless of the layer height you print with. And that first layer, because it is squished onto the build plate, is not guaranteed to be 0.2 mm either.
* In the case of horizontal holes, the edge of the perimeter, due to line width and printing outer perimeters last, may intrude a bit into the hole, and at the bottom of the hole this intrusion can result in an entire layer's worth of vertical clearance loss, which is why it's a good idea to put a 1-layer-deep point at the bottom of a teardrop.

These variations typically combine to cause 0.1 mm clearance horizontally to be a pretty tight fit. I find that 0.2 mm clearance for small holes is pretty snug. And the vertical variations are why the teardrop needs a little point at the bottom.

Which brings me to your question about what's wrong with 3 layers of bridging. More than 3 layers would likely work fine if the layer height in the design equals the layer height being printed. It may not be in exactly the right place due to the first layer thickness offset, but close enough to work. Where it goes awry is when you have the design layer height smaller than the print layer height, in which case you'll get the first layer of bridging and maybe the second, but likely not the third. To compensate for this, when I release a design to the public that has holes, I assume a layer height of 0.3 mm because printing at a smaller layer height is going to hit all the bridges, and may print two layers on one bridge, but that's OK.

_Originally posted by @amatulic in https://github.com/BelfrySCAD/BOSL2/issues/1467#issuecomment-2540528348_

adrianVmariano, 2024-12-13:
The test fit device I was using is for vertical holes.   For me a 0.5mm clearance sounds ridiculously large unless what you want is not a "fit" so much as a roll around.  I have not attempted any kind of test fit on horizontal holes.  But I have found horizontally printed screw holes to work fine at M3-M5 size without the bottom point and without a huge clearance.  But I am not sure what clearance I was using.  

If the focus is on tiny holes, though, there's another issue.  In OpenSCAD a small hole doesn't have resolution to resolve those points, generally speaking, unless you've gone crazy with your `$fa` and `$fs` settings.   Or on the flip side, the point will be there automatically because of faceting.   Actually it seems like using `circum=true` might be more important and more useful than adding a bottom corner, especially if the focus is on tiny holes.  It appears that screw_hole is not using `circum=true`, which is maybe a mistake.

amatulic, 2024-12-13:
Design guidance I was referring to in #1467 - https://www.hydraresearch3d.com/design-rules#holes-horizontal

RAMilewski, 2025-02-03:
If you're going to break open the code for teardrop anyway:

There are situations where it would be useful to be able to apply the teardrop rounding to the top of a cuboid.

Here's a shell for a prototype USB dongle made by splitting the box and rotating the top half for printing.   I would have preferred to use rounding rather than chamfering, but that would require a teardrop rounding on the top to print well.

[image]

[image]

adrianVmariano, 2025-04-27:
@amatulic What's the conclusion on this issue.  Does teardrop need to have a bottom corner option?  This feels like something that appears trivial but would actually be tricky.

amatulic, 2025-04-27:
@adrianVmariano I thought I made my view clear.

I wouldn't use the BOSL2 teardrop for horizontal holes for 3D printing because it doesn't conform to the guidance I use to make small horizontal holes. I don't believe the bottom corner should be an option; rather, it should always be there, and both bottom and top should be dependent on a layer height parameter. The bottom corner would be invisible for large holes, and makes things fit better in small holes. It's common to use filament as a hinge pin, and the bottom corner is necessary for that tiny-size 2mm hole if it's horizontal.

@RAMilewski made a point about rounding in general, which is indirectly related to teardrops. Whenever I make a rounded design for 3D printing, I also cannot use BOSL2's rounding features. For printing bottom-rounded edges, the rounding required has a 40° tangent to the build plate. Basically the rounding curve must be positioned so that it is cropped when the tangent angle is 40° from horizontal.

adrianVmariano, 2025-04-27:
I don't understand your second issue:  you can't use the existing rounding because you need 40 degrees instead of 45 degrees?  It seems like 40 degrees is more challenging to print, so 45 deg should be fine.   I think this is totally different from the issue reported by  @RAMilewski which is that he wants the teardrop on the top instead of the bottom.   (Also note that teardrop can in principle support any angle...but there's no way to pass that parameter to most of the things that do teardrop rounding.)  

We can't make teardrop always produce a bottom corner because that would break teardrop corner rounding.  There is also a problem with parameter propagation, which admittedly already exists, but it would be good to work out a scheme were horizontal holes can be produced by other modules that call teardrop and produce the desired results.   Having a dependency on a parameter (layer height) is an added complication.

amatulic, 2025-04-27:
40° overhangs aren't challenging to print. Violating the 45° rule by 5° is something that no printer has trouble with. Some printers can go further but 40° is save.

Parameter propagation shouldn't be an issue if layer height defaults to zero, then nothing else that uses teardrop needs to change. The dependency on layer height is mandatory for printing a horizontal hole reliably, and 0.3 mm is usually a safe bet even for 0.2 mm layers.

adrianVmariano, 2025-04-27:
My point is that I'm puzzled by your saying you can't accept the easier-to-print 45 deg angle that current BOSL2 rounding produces and that you can't use it because you **require** a harder-to-print 40 deg angle instead.   I'll note also that there are various BOSL2 rounding methods to produce teardrop rounding with the specified teardrop angle---I think `edge_profile_asym()` should be able to do that.  

I've printed horizontal holes with the existing code, so just how mandatory the corner is---that's not so clear.  We can't have a `$layer_height` parameter and tell people to set it to 50% more than the layer height.  Presumably there needs to be a parameter specifically linked to teardrop.  And there's also the teardrop angle as a parameter.   And it's not exactly clear that for a large hole, a single layer height is enough at the top.  Can you really get away with a single parameter there?  And there's the vague direction to use double the layer height when the layer height is "small".  So the whole business is pretty hazy, in fact.

amatulic, 2025-04-28:
I don't know everything in BOSL2, so if there's a way to get the bottom angle I want on a rounded curve, that's great.

How small do you print holes? Try printing one to fit a 1.75 mm filament or even a 3 mm screw with a typical line width setting of 10% above nozzle diameter. The bottom corner is needed then. And if it's always there, it isn't noticeable on larger holes, because it's only one layer difference at the bottom, falling within the layer rounding error for tops of surfaces.

Yes, you can get away with a single parameter. The double layer height doesn't matter for the bottom corner, only for the larger tip of the teardrop, and then only if you're truncating that tip to a trapezoid that forms a bridge instead of having a pointy tip. It matters when the layer height is small (like less than half the nozzle diameter) because bridges come out of the nozzle thicker than that, so you need a taller height to fit the bridge.

My experience, for a 2 mm horizontal hole, is that a single layer space above the hole is fine when printing at 0.3 mm. I get the correct clearances for in that case. That designed-in layer height would work for printing smaller layers too. If the designed-in layer height is 0.3 mm, that is equivalent to two layers at 0.15 mm, which also works well.

adrianVmariano, 2025-04-28:
```
include<BOSL2/std.scad>

$fs=.1;$fa=1;
cuboid(10, rounding=3, teardrop=true);
```

[image]

Also Example 8 and after here:

https://github.com/BelfrySCAD/BOSL2/wiki/rounding.scad#functionmodule-offset_sweep

As I said, I **have** printed a 1.75mm hole for a filament hinge pin on a 9" long box without using a bottom corner.  

The teardrop capability is not a "horizontal hole" function, it is a geometrical object that can be used to make horizontal holes.  The corner **cannot** always be present on teardrops because people may be using the teardrop for other purposes.  It is not acceptable to add geometrical deviations because "they wont be noticed".   That corner could actual be responsible for a print failure by elevating a part of the model one layer line above the base.  (I had this happen where differing faceting resulted in the "base" of my model being on two levels, which lead to very mystifying bed adhesion issues.)  

Also broadly speaking, users need to be able to do what they want with their geometry, not what you think they should want.  

With regards to using the same parameter at both ends, you're saying that if I want a 2 inch hole it suffices to put the top flat on a teardrop 1 layer line above the tangent?

adrianVmariano, 2025-04-28:
Do you mean this?  

[image]

Never have seen that or tried it.  Gets complicated though if you want the edge treatments to be the same size.  Why do you prefer it to the teardrop rounding?  

Teardrop **is** intended to be used for horizontal holes and is in fact used for them in various places in the library.   But it's also a teardrop shape that could be used for anything.  It's a bit unclear that a separate "horizontal hole" function is desirable.  

Keep in mind that you're living in a mindset that is very limited.  What if I'm not going to 3d print but have some other use for the model?  What if I have soluble support materials and don't care about unsupported regions.   There's no reason that horizontal holes need to be teardrop shaped generally.  A user may also make a model where the holes that are horizontal in the model are actually printed vertically.  In other words, it makes more sense to specify a "teardrop" hole which clearly identifies what will happen than to flag holes as "horizontal" and have that make them into teardrops.

amatulic, 2025-04-28:
I prefer that to a teardrop rounding because it gives a more rounded look without introducing flat surfaces on the rounded edge. I also found that this sort of rounding, done on both top and bottom, is more comfortable to hold when I designed these pliers because the round surface meeting the edge of the handle feels less sharp than a flat surface meeting the edge. For a symmetrical look, I always round the top and bottom of my objects like this.

[image]

Another example is this plug cover I designed. It must be printed vertically on the short edge, so all four edges of the plate are rounded so that the bottom edge rounding meets the build plate at 40°. The result here is the part appears nicely rounded with no flats in the rounding before it meets the edges. It's hard to tell in the picture due to the brightness of the white. Not only is the large plate rounded toward all four edges, but the two vertical sides are also rounded toward the short sides, resulting in clean curves and corners with no layer artifacts. This just wouldn't look good with teardrop rounding.

[image]

If it's too complicated to introduce features into the teardrop to make it into what I consider a useful horizontal hole profile, particularly if it's used for other purposes like rounding, then it might be useful to have a horizontal hole model to complement the unsupported vertical hole model. As I said, the BOSL2 teardrop just doesn't work for me, so I don't use it. I'll continue to use the horizontal hole profile as shown in that guidance document I linked in my first comment above -- and that isn't the only place I've seen similar guidance.

Equal extents on the top and bottom have served me well, but then, admittedly, I haven't needed to print horizontal holes that are bigger than 6mm. Any bigger hole I've needed, I've contrived to print it vertically. The only attempt I made to do a large hole horizontally was with [this King Kamehameha I spear head replica](https://www.printables.com/model/129111-king-kamehameha-i-spear-head) I did in Tinkercad as a costume prop (needed semicircle holes) and I found myself reaming out the bridges in the holes to make the parts fit together, as well as filing off the top layer of the half-cylinders that fit into the hole. This wasn't a problem with the [Gugnir spear head](https://www.printables.com/model/129128-odins-spear-gungnir) I did in OpenSCAD a few years later, but then I made sure to design the cylindrical inserts to omit that interfering top layer.

adrianVmariano, 2025-04-28:
Really?  A round surface meeting an edge feels smoother than a flat surface meeting an edge?  Surely this is determined by the angle of the edge, not by the shape of the curve as you move away from the edge.   I guess I'll have to do some test prints to see what I think.  

It seems like it shouldn't be hard to add support for this in the masking code and in offset_sweep.  We just need a parameter name that specifies the angle.  Hmmm.  Maybe we also need the option to specify the rounding by height.    Constructing the profile manually:

```
$fa=1;$fs=.1;
diff()
  cuboid(10)
    edge_profile_asym([TOP,BOTTOM])
      back_half(planar=true,y=-.01)fwd(3*(1-cos(40)))mask2d_roundover(r=3);
```

[image]

Another problem I noticed is that the `corner_type` parameter to edge_profile_asym() only works for negative roundings but if you want to round the side edges of this cube you need it here.  So it should probably be enabled to work always.  Then you could add side roundings that would blend.  

Adding support for this kind of rounding directly to cuboid() may not be so simple.  I don't know because I have no clue how cuboid works.  I think cyl() makes its roundings by creating a profile internally, so that could conceivably be adjusted to do this thing as well.   Maybe `clip_angle=40`?  

You can of course do this by making your model with full rounding and then shaving off the bottom.   

Introducing additional features into teardrop is only complicated, really, in that it will probably take 2 hours to figure out the code to add that bottom point.   The complexity I'm really talking about is how to control teardrop parameters when teardrop is called indirectly.  And it's not complexity so much as API clutter that I'm concerned about here.  We can create 17 global `$` variables to control ever nuance of teardrop shape and that's not "complex", just ugly.   The question is how to minimize that.   Adding a 3rd hole type is **not** a simplification!  That means everybody making holes now has to provide **three** options.  That's definitely complex.

amatulic, 2025-04-30:
> A round surface meeting an edge feels smoother than a flat surface meeting an edge? Surely this is determined by the angle of the edge

One would think so, and I admit it's subtle. But you can sense a flat part on both sides of the edge and your brain says "corner". Look at the picture of the pliers above. The rounding is maximized on both sides so the roundings meet in the middle, with no flat area. Imagine how tight the curved outside of the handle would have to be if there had to be a flat surface tangent to the curve that meets the edge at 40° from horizontal. In the case of the plug cover picture, it's more a matter of appearance; it looks better if the curve ends at an edge instead of ending at a tangent flat part that goes on to meet the edge.

> You can of course do this by making your model with full rounding and then shaving off the bottom.

And I have done this, but usually I just construct the curve I want. Calculating the angle of a circle clipped at the bottom so the arc starts out 40° from horizontal isn't difficult to do. Funny, while I have my own library function for a horizontal hole, I never actually put this rounding in a library, I always just write the code over again to calculate the position of the circle center to start the arc at -50° so the arc angle at the start is 40°.

amatulic, 2025-05-04:
Should this issue be closed? The rounding that starts at an angle has been implemented.

As for the teardrop having an adjustable angle and gap at the trapezoid top and a 1-layer-deep corner at the bottom, it sounds like changing the teardrop isn't practical due to complexity of the API and other things that depend on it. If it's undesirable to have a horizontal hole profile that conforms to the filament printing design guidance I described above, it's OK to close this as "won't do" or "not planned."

adrianVmariano, 2025-05-04:
I am not sure.  @revarbat seemed to be opposed to the bottom corner as unnecessary.  

The API concerns I raised somewhat exist anyway.  Adding more options to teardrop just makes them more obvious.   As I mentioned before, the real challenge from my perspective is actually implementing a bottom corner in a way that doesn't disturb the faceting.   I suppose a hack like making a no-corner teardrop and then cutting it to add the teardrop could be a solution.  

But I wanted Revar on board with the bottom corner before doing this.

revarbat, 2025-05-04:
I'm not actually against the bottom corner, as it is a fixed sized feature based on layer height that `circum=` can't quite guarantee.

closed 2025-05-06 by adrianVmariano