BelfrySCAD/BOSL2 #1539: Update rotate_sweep to match changes to rotate_extrude

adrianVmariano, 2025-01-06:
```
First, I changed it so that you can start an extrusion anywhere around
the circle you want to, using the "start" parameter.

    rotate_extrude(angle=90, start=45) square(10);

By itself, this is a small thing; you can achieve the same result by
slapping a rotate() around your extrusion.  However, it sets the stage
for the second change...

Second, I made it so that "start" defaults to zero if you specify
"angle", and defaults to 180 if you do not.  This means that all
extrusions that specify "angle"[*], specifically including those that
specify angle=360, start on the +X axis, to the right of the origin.
```

I wonder if this change will break anything.

amatulic, 2025-01-06:
If you omit 'angle' it defaults to 360. If I'm working with arrays of vertices rather than the builtin OpenSCAD geometries that nobody has access to, I would expect the zero vertex to be on the positive X axis always, even if I don't specify angle. What would be the point of putting it on the negative X axis?

adrianVmariano, 2025-01-06:
According to Jordan (I didn't verify) if you don't give an angle or give angle=360, then it starts on the X- axis.  Otherwise, it starts on the X+ axis.  That's the current behavior.  

We would like for `rotate_sweep(path,args);` to give the same result as `rotate_extrude(args)polygon(path);`

adrianVmariano, 2025-03-04:
Sample code:

```
color("red") cylinder(h=1,r=10, $fn=3);
color("green") translate([0,0,1]) rotate_extrude($fn=3) square([10,1]);
```

adrianVmariano, 2025-03-05:
As it happens, rotate_sweep already always starts on the X+ axis, so it's inconsistent with rotate_extrude.   Added the angle argument.

closed 2025-03-05 by adrianVmariano