BelfrySCAD/BOSL2 #1545: [BUG] bezpath_closest_point() not working

dan-p3rry, 2025-01-18:
**Describe the bug**
The closest point is nowhere close to the point it should be.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/beziers.scad>

bez = [[-78.2772, 0], [-78.2772, 45], [-78.2772, 90], 
    [0, 90], 
    [78.2772, 90], [78.2772, 45], [78.2772, 0]];

path1 = bezpath_curve(bez, 17);

midpt = bezpath_closest_point(bez, [0, 90], 3, 5);

echo(path1=path1);
echo(midpt=midpt);
```

**Expected behavior**
The closest point should be at or near [0, 90] , the same value as the second argument passed to bezpath_closest_point().  Actual value is  [1, 0.00347222], which maybe by coincidence is roughly equidistant to all points on the bezpath.
 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

- OpenSCAD Version: Jan 4, 2025 download
- Other libraries used:

amatulic, 2025-01-18:
By the way, it is no longer necessary to include beziers.scad; it's now part of std.scad.

dan-p3rry, 2025-01-18:
What's the effective date for not requiring beziers.scad?  It was required for this Jan. 4 download.  Is there a release history somewhere?  Thanks!

amatulic, 2025-01-18:
The bug is where bezpath_closest_point returns `[min_seg, bezier_closest_point(curve, pt, max_err=max_err)]]` which isn't a coordinate; `min_seg` is just an index value, and `bezier_closest_point()` doesn't return a coordinate, it returns a `u` bezier parameter that can be used in `bezier_points()` to return an actual coordinate. I tried to fix this but now I'm getting the result in the wrong quadrant for some reason.

Under the 'code' tab on GitHub, see the comment next to std.scad; beziers.scad was included in std.scad as of 2 weeks ago. It does not harm to have a redundant include, however. I'm relatively new participating here, but my understanding is that while BOSL2 is still a 'beta' there are no formal release notes or versioning history. There is a PR about that.

amatulic, 2025-01-18:
Ah. I had a typo in my tests. It's working now. I'll submit a correction.

amatulic, 2025-01-18:
Umm.... actually I'm going to cancel this PR. @dan-p3rry we both need to RTFM. It's working properly. According to the documentation,`bezpath_closest_point()` is *supposed* to return `[segnum, u]` for the closest position on the bezier path to the given point.

You can't use the return value directly as a coordinate, you need to pass `segnum` and `u` into `bezpath_points()`. See documentation: https://github.com/BelfrySCAD/BOSL2/wiki/beziers.scad#function-bezpath_points

amatulic, 2025-01-18:
@dan-p3rry because I didn't create this issue, I can't close it, but you can.

dan-p3rry, 2025-01-18:
OK, I see that.  I guess I'd like to change this from a bug report to enhancement request then.  

In my sample above, path1 output goes right through my 4th (middle) point at [90, 0], and shows up in the echo() statement.  
But bezpath_closest_point() uses splicesteps=16 (an even number) so it misses the center point.  My enhancement request would be to be able to define splinesteps for bezpath_closest_point(), or to have an option to output the midpoint of the bezpath.

dan-p3rry, 2025-01-18:
BTW, the bezpath is symmetric around y axis, but the result of bezpath_closest_point is ECHO: xy = [0.81256, 89.9984].

amatulic, 2025-01-18:
That's because you set a `max_err` value of 5, so it's going to return the first thing it finds within that error bound.

amatulic, 2025-01-18:
Can you create a new issue for the enhancement?

dan-p3rry, 2025-01-18:
> That's because you set a `max_err` value of 5, so it's going to return the first thing it finds within that error bound.

You are correct.  That large value was from the beginning when I didn't rtfm.  No need for enhancement request then.  Thanks for your help.  Closing the issue.

closed 2025-01-18 by dan-p3rry