BelfrySCAD/BOSL2 #1547: [BUG]2 typos in file Beziers_for_Beginners.md

WernerP, 2025-01-22:
**Describe the bug**
There are 2 types in file "Beziers_for_Beginners.md" with the address
https://github.com/BelfrySCAD/BOSL2/blob/8f795bb93b2167d75b2167bf225aef52c3d4f540/tutorials/Beziers_for_Beginners.md

in "Bézier Paths"
the line bez = [[0,0], [30,30], [0,50], {70,30]  [0,100]];
2 typos: { before 70,30] and , is missing after 70,30]

RAMilewski, 2025-01-22:
Not sure what's happening here.  The version in the wiki appears correct.

https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Beziers_for_Beginners

WernerP, 2025-01-23:
> Links:

Good morning,

the same 2 typos are in the file Beziers_for_Beginners.md in the install 
directory of BOSL2. On my machine its in
OpenSCAD\libraries\BOSL2\tutorials

The file Beziers_for_Beginners.md has been created / modified at 09. 
Jan. 2025 on my machine

The examples in this file and on the address 
https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Beziers_for_Beginners
seems to be different

On 
https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Beziers_for_Beginners 
there is no typo

In the above mentioned file I see

----- Begin of block
The list of control points for a Bézier is not an OpenSCAD path. If we 
treat the list bez[] as a path we would get a very different shape.  
Here the Bézier is in green and the OpenSCAd path through the control 
points is in red.

```openscad-2D
include<BOSL2/std.scad>

bez = [[0,0], [30,30], [0,50], {70,30]  [0,100]];
debug_bezier(bez, N = 2);

While the bez variable in these examples is a list of points, it is not 
the same as an OpenScad path, which is also a list of points.  Here we 
have the Bézier curve shown in green and an OpenSCAD path through the 
same point-list in red.

```openscad-2D
include<BOSL2/std.scad>

bez = [[20,0], [60,40], [-20,50], [25,80]];
debug_bezier(bez, N = 3);
color("red") stroke(bez);
```
----- End of block

I hope this helps

Bye

amatulic, 2025-01-23:
It seems the source file https://github.com/BelfrySCAD/BOSL2/blob/master/tutorials/Beziers_for_Beginners.md was changed in the repo since the last time the Wiki tutorial was built from it. I have submitted a PR to make the corrections, and I also found other typos and minor grammar things to fix. I can't merge the changes or rebuild the tutorial; that would be up to @revarbat

commit 4ff847ab (2025-01-24): Merge pull request #1551 from amatulic/anachronist_tutorials

Fix #1547 - typos in example, minor spelling and grammar improvements
commit c67b7b56 (2025-01-24): Fix #1547 - typos in example, minor spelling and grammar improvements