BelfrySCAD/BOSL2 #1548: [BUG] round3d() in combination with any form of transform along Y

yahbluez, 2025-01-22:
round3d() acts wrong full if combined with translations.

This cide snip shows the effect:

round3d(or=2) fwd(20) cube([20, 100, 10], );
up(10) cuboid([20, 100, 10]);

expected is that booth cubes have the same size, 
but the rounded one is shorter by the amount the cube is moved.

[image]

Used the nightly openscad snap and cloned the BOSL2 today to ensure the bug is i the newest version.

amatulic, 2025-01-22:
The `fwd()` isn't relevant. `round3d()` still truncates the cube without it. Here are two identical cubes overlaid, one rounded, one not.

[image]

````
include<BOSL2/std.scad>
round3d(or=2) cube([20,100,10]);
#cube([20,100,10]);
````
The rounded cube has length 50 regardless of the original cube's y dimension is set.

**It works only as long as the y dimension is less than 50.** It seems like a hard-coded limit somewhere but I haven't looked at the code yet.

yahbluez, 2025-01-22:
Thank you @amatulic that's one step more to get that fixed.

amatulic, 2025-01-22:
It turns out all dimensions get truncated. Investigating further....

If you set the parameter `size=300` in your call to `round3d()`, your example works. Basically set `size` to *more than* double the amount your object extends from any axis. This isn't documented.

`round3d()` is in miscellaneous.scad. It's a small module that simply calls `offset3d()` (in the same file) three times in a nested fashion. `round3d()` has a default parameter `size=100`, which gets passed to `offset3d()`, and this is the culprit. The `size` parameter is used in a `difference()` operation with a centered cube, which would explain why a non-centered cube truncates at 50, because a size 100 centered cube extends only out to 50 on each axis.

I note the documentation says:
> This is an *extremely* slow operation.  I cannot emphasize enough just how slow it is.  It uses `minkowski()` multiple times. Use this as a last resort.  This is so slow that no example images will be rendered.

Based on that, I suspect it wasn't tested extensively either.

It probably wouldn't hurt to make the default `size` something bigger, like 1000.

yahbluez, 2025-01-22:
@amatulic 

Thank you again, i did a work around for the model i was working on.

To put some shame on me, i did not do a <F12> "Go to definition" in code.
If i had done  so i would have seen the size parameter in the first line of the definition.

That teaches me to look into modules that bother.

[image]

That is what i made today:

https://makerworld.com/en/models/1026215#profileId-1008411

adrianVmariano, 2025-02-23:
Documented size and made default large, which doesn't appear to cause issues.

closed 2025-02-24 by adrianVmariano