BelfrySCAD/BOSL2 #1549: [BUG] Error calling vnf_bend() twice on a hulled torus

AlecMRogers, 2025-01-23:
**Describe the bug**
When I execute code designed to produce a warped disc, I get warnings and a solid that is missing many faces.

**Code To Reproduce Bug**
$fa=4;
$fs=4;

include <BOSL2/std.scad>
include <BOSL2/polyhedra.scad>

vnf_polyhedron(
    vnf_bend( rot([0.0,0.0,90.0], 
        p=vnf_bend( move([0,0,20],
            vnf_hull( torus(d_maj=11.0, d_min=5.0))) ,
        r=20, axis="Y" )),
    r=9, axis="Y" )
);

**Expected behavior**
A rendered solid with correct VNF representation.

**Screenshots**
[image]

**Additional context**
[OpenSCAD](https://www.openscad.org/) version 2024.10.28 (git 7f05c6294)

Relevant Warnings (?):
[WARNING: undefined operation (undefined > number) in file Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/BelfrySCAD/BOSL2/issues/1134,/Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad)
[WARNING: undefined operation (undefined > number) in file Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/BelfrySCAD/BOSL2/issues/1134,/Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad)
[WARNING: undefined operation (undefined > number) in file Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/BelfrySCAD/BOSL2/issues/1134,/Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad)
[WARNING: undefined operation (undefined > number) in file Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/BelfrySCAD/BOSL2/issues/1134,/Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad)
[WARNING: undefined operation (undefined > number) in file Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/BelfrySCAD/BOSL2/issues/1134,/Users/arogers/Documents/OpenSCAD/libraries/BOSL2/vnf.scad)

adrianVmariano, 2025-01-25:
The problem is that the first vnf_bend() is producing output that has non-coplanar faces.  A work-around is to call vnf_triangulate on the vnf_bend() output.  

It seems like maybe we need to triangulate the output of vnf_bend?  I'm not sure if there's a better solution.

AlecMRogers, 2025-01-25:
I don’t think the bend algorithm needs to touch the faces at all (except in degenerate cases); in this case, it should just be modifying two coordinates (e.g. X, Z for a bend on Y) of all points.Sincerely,-alechttps://GnosticModels.com508-361-2039

adrianVmariano, 2025-01-25:
vnf_bend() subdivides all the faces and then recomputes the coordinates as you say.  

Even without the subdivision, a face that has more than 3 points can become non-coplanar under the remapping of the vertex values.

AlecMRogers, 2025-01-26:
I don’t know what policy is on non-triangular faces in general, but it seems safer to call vnf_triangulate() on the input to vnf_bend() (i.e. as opposed to adding it after the bulk of the algorithm).

Thank you.

adrianVmariano, 2025-01-26:
I'm not sure I understand.  You're calling vnf_bend twice.  The problem occurs because vnf_bend can produce noncoplanar faces in the output, which is a bug.  You can work around this at the moment by invoking vnf_triangulate after, not before, vnf_bend. 

If you triangulate before vnf_bend it probably won't help because vnf_bend slices the vnf with a vnf_slice call and that turns triangles into quadrilaterals, which can then end up non-coplanar.  The only places that make sense are either immediately after the vnf_slice call or at the end of vnf_bend.  It's probably better to do it right after the slice, before the bend because the triangulation code might not like non-coplanar quadrilaterals.  

The general situation with non-triangular faces is confusing.  I would say that the short answer is there has been some concern in the past that CGAL would throw errors when faces weren't triangular which lead to an aggressive triangulation practice in the library.  However, for many operations that we want to perform on VNFs it's actually better when they are not triangulated.  I wrote an untriangulator because of this.  In fact, triangulating before slicing produces nasty artifacts in textures and I ended up de-triangulating all the built-in textures to try to improve that.

AlecMRogers, 2025-01-26:
If nontriangular faces are a legitimate thing in vnf in general, than any nontriangular input to vnf_bend will produce the problem, not just output from vnf_bend itself.

adrianVmariano, 2025-01-26:
Yes, that's true.  But I'm not sure what your point is.  Non coplanarity doesn't occur until the vertices are remapped, which is the last thing that happens in vnf_bend.

adrianVmariano, 2025-01-26:
Note: non-triangular faces are a legitimate thing, but all legit faces must be coplanar.

closed 2025-01-26 by adrianVmariano