BelfrySCAD/BOSL2 #1556: [BUG] use bosl2 is broken in newer versions

smartmeta, 2025-01-29:
**Describe the bug**

Use does not work for openscad verisons newer than 2021.01

I tried all my combinations in these versions
[image]
(Did some additional inbetween.)

The error message is:

```
Compiling design (CSG Tree generation)...
[WARNING: Ignoring unknown variable '$anchor_override' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3087](https://github.com/BelfrySCAD/BOSL2/issues/3087,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[WARNING: Ignoring unknown variable '$attach_to' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3746](https://github.com/BelfrySCAD/BOSL2/issues/3746,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[WARNING: Ignoring unknown variable '$transform' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/transforms.scad, line 1614](https://github.com/BelfrySCAD/BOSL2/issues/1614,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/transforms.scad)
[WARNING: undefined operation (undefined * vector) in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/transforms.scad, line 1614](https://github.com/BelfrySCAD/BOSL2/issues/1614,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/transforms.scad)
[WARNING: Ignoring unknown variable '$tags_shown' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 4257](https://github.com/BelfrySCAD/BOSL2/issues/4257,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[WARNING: Ignoring unknown variable '$tags_shown' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 4257](https://github.com/BelfrySCAD/BOSL2/issues/4257,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[ERROR: Assertion '(is_list($tags_shown) || ($tags_shown == "ALL"))' failed in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 4257](https://github.com/BelfrySCAD/BOSL2/issues/4257,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[TRACE: called by '_is_shown' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3110](https://github.com/BelfrySCAD/BOSL2/issues/3110,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[TRACE: called by 'if' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3110](https://github.com/BelfrySCAD/BOSL2/issues/3110,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[TRACE: called by 'children' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/transforms.scad, line 1615](https://github.com/BelfrySCAD/BOSL2/issues/1615,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/transforms.scad)
[TRACE: called by 'children' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/builtins.scad, line 30](https://github.com/BelfrySCAD/BOSL2/issues/30,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/builtins.scad)
[TRACE: called by 'multmatrix' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/builtins.scad, line 30](https://github.com/BelfrySCAD/BOSL2/issues/30,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/builtins.scad)
[TRACE: call of '_multmatrix(m = [[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]])' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/builtins.scad, line 30](https://github.com/BelfrySCAD/BOSL2/issues/30,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/builtins.scad)
[TRACE: called by '_multmatrix' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/transforms.scad, line 1615](https://github.com/BelfrySCAD/BOSL2/issues/1615,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/transforms.scad)
[TRACE: call of 'multmatrix(m = [[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]])' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/transforms.scad, line 1609](https://github.com/BelfrySCAD/BOSL2/issues/1609,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/transforms.scad)
[TRACE: called by 'multmatrix' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3101](https://github.com/BelfrySCAD/BOSL2/issues/3101,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[TRACE: call of 'attachable(anchor = [0, 0, 0], spin = 0, orient = [0, 0, 1], size = [50, 50, 10], size2 = undef, shift = undef, r = undef, r1 = undef, r2 = undef, d = undef, d1 = undef, d2 = undef, l = undef, h = undef, vnf = undef, path = undef, region = undef, extent = true, cp = [0, 0, 0], offset = [0, 0, 0], anchors = [], two_d = false, axis = [0, 0, 1], override = undef, geom = ["prismoid", [50, 50, 10], [50, 50], [0, 0], [0, 0, 1], function(anchor) [undef, undef, undef], [0, 0, 0], [0, 0, 0], []], expose_tags = false, keep_color = false)' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/attachments.scad, line 3078](https://github.com/BelfrySCAD/BOSL2/issues/3078,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/attachments.scad)
[TRACE: called by 'attachable' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/shapes3d.scad, line 394](https://github.com/BelfrySCAD/BOSL2/issues/394,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/shapes3d.scad)
[TRACE: called by 'if' in file ../../../../../../../Documents/OpenSCAD/libraries/BOSL2/shapes3d.scad, line 350](https://github.com/BelfrySCAD/BOSL2/issues/350,/Users/oho/Documents/OpenSCAD/libraries/BOSL2/shapes3d.scad)
```

**Code To Reproduce Bug**

My setup has 6 scad files and 2 scad-lib-files

[image]

Here is my test setup:

File 't_direct_include'
```
include <BOSL2/std.scad>
cuboid([50, 50, 10], chamfer = 0.3);
```

This allways works.

File 't_direct_use'
```
use <BOSL2/std.scad>
cuboid([50, 50, 10], chamfer = 0.3);
```

This did work in v2012.01, but not in the following versions.

----
Now the same with lib-files.

File 'lib_include.scad'
```
include <BOSL2/std.scad>
module strange (width, length, height) {
    cuboid([width, length, height], chamfer = 0.3);
}
```

File 'lib_use.scad'
```
use <BOSL2/std.scad>
module strange (width, length, height) {
    cuboid([width, length, height], chamfer = 0.3);
}
```

And the four options to mix them:

File 't_include_lib_include.scad'
```
include <./lib_include.scad>
strange(50, 50, 10);
```

File 't_use_lib_include.scad'
```
use <./lib_include.scad>
strange(50, 50, 10);
```

File 't_include_lib_use.scad'
```
include <./lib_use.scad>
strange(50, 50, 10);
```

File 't_use_lib_use.scad'
```
use <./lib_use.scad>
strange(50, 50, 10);
```

In the 2021.01 version all combinations work. In later versions only works when the lib-file is included and also BOSL2 is included.

**Expected behavior**

Also newer versions should allow to "use" BOSL2.

**Additional context**
I'm on macOS with AppleSilicon, but I guess that is not the cause.

adrianVmariano, 2025-01-29:
You can never include BOSL2 using "use".   Due to the performance problems with "use" we do not support it.

closed 2025-01-29 by adrianVmariano

smartmeta, 2025-01-29:
`You can never include BOSL2 using "use". `

Sure you can in version 2021.01.

'Due to the performance problems with "use" we do not support it.'

Then you should mention this in the documentation (somewhere at the top).

adrianVmariano, 2025-01-29:
No, you cannot.  There are things we have done in BOSL2 that simply will not work if the library is "used".  Changing the library to work with "use" would be a nontrivial effort. 

There is no way we can make use of "use" not display errors.   (And why would that be desirable: it's an error and won't work right.)

Every single example that shows how to use BOSL2 uses "include" throughout all of the documentation.  It's not clear where a warning to avoid "use" would go that would be useful.

revarbat, 2025-01-29:
> You can never include BOSL2 using "use".   Due to the performance problems with "use" we do not support it.  

Also, `use <>` *cannot* (by design) provide constants like `UP` or `INCH`.

revarbat, 2025-01-29:
> It's not clear where a warning to avoid "use" would go that would be useful.

We'll, at the top of every file, we could have a line reading:

>**Warning: Do not import BOSL2 using `use<>`.** It will fail catastrophically. We don't support it due to performance bugs in OpenSCAD's implementation of `use<>`.

revarbat, 2025-01-29:
In any case, at this point, we'd have to untangle a terrible web of calls that were placed in files based on topic, rather than based on the call graph.

amatulic, 2025-01-29:
@revarbat and @adrianVmariano - this 'use' versus 'include' distinction hit me also the first times I started dabbling with BOSL2. My habit is to use 'use' always for my own libraries that I have written, and I'm sure that's the habit of others. The reason to use 'use' is that you can include executable demos right in the file that can be previewed in that file, but when you 'use' the file, you get access to the functions and modules and the demo doesn't run. That's how I've been distributing my own code.

It wouldn't hurt to include a note about this distinction, and the main reason for it (constants don't work with 'use'), in README.md.

adrianVmariano, 2025-01-29:
I would say that we could work around the constants thing somewhat by having a top level file that must be included that includes the constants and then uses everything else.  The thing we can't work around are the potentially catastrophic performance penalties associated with "use", where any time you call a function in a file it evaluates all assignments in the top level of that file.