BelfrySCAD/BOSL2 #1563: Review validity of vnf_validate() error conditions

amatulic, 2025-02-13:
Regarding https://github.com/BelfrySCAD/BOSL2/wiki/vnf.scad#module-vnf_validate

Two conditions should be reviewed, and removed if not applicable:

* BIG_FACE warning: This has **never** been an issue in the years I've been using OpenSCAD, starting with the 2015 version. Before I came across BOSL2, I released several polyhedron-based designs made completely from nonplanar quads ([example](https://www.printables.com/model/159163-elliptical-blade-naca-airfoil-propeller-library), [example](https://www.printables.com/model/459462-over-engineered-jst-connector-pliers)). **Not once** have I ever encountered a rendering error or CGAL complaint due to a nonplanar face. If it's nonplanar, it gets triangulated for you.
* MULTCONN error: For the example given, with an edge sharing four faces, neither CGAL nor Manifold complains, both render the object.
  * The STL file output simply has 24 triangle faces in it, that's all. When previewing in 3D Builder, it's a valid STL, regardless of whether it was generated by Manifold or CGAL.
  * In both cases, the 3MF file output *does* cause 3D builder to complain and offer to repair it.

Perhaps that difference between STL and 3MF should be reported to the OpenSCAD repo. I thought 3MF was essentially an extension of STL.

adrianVmariano, 2025-02-13:
I don't think you can make a valid object with CGAL that has an edge sharing four faces.   Maybe it's possible with manifold?  

The big face thing is somewhat confusing.  I questioned that, but I think we did have a bug recently where we had to run vnf_triangulate because a large face resulted in a rendering error.

amatulic, 2025-02-13:
I tried it with both CGAL and Manifold. It worked in both cases, both exported a valid STL, but neither exported a valid 3MF.

adrianVmariano, 2025-02-13:
Did you have a second shape in your model?

adrianVmariano, 2025-02-13:
Unfortunately, I don't remember what the situation was with the big face.

adrianVmariano, 2025-02-13:
```
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 
Rendering cancelled.
Geometries in cache: 4
Geometry cache size in bytes: 111348704
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:02:06.971
WARNING: No top level geometry to render 
```

adrianVmariano, 2025-02-13:
```
v=vnf_join([
           cube(10),
           move([10,10], cube(10))]);

vnf_polyhedron(v);
sphere(4);
```

amatulic, 2025-02-13:
This has no problem with CGAL or Manifold:
````
include <BOSL2/std.scad>
vnf = vnf_triangulate(linear_sweep(union(square(50), square(50,anchor=BACK+RIGHT)), height=50));
vnf_polyhedron(vnf);
````
...and exporting the STL gives me the correct shape.

However, for this:
````
include<BOSL2/std.scad>
v=vnf_join([
    cube(10),
    move([10,10], cube(10))
]);
vnf_polyhedron(v);
sphere(4);
````
I get an error with CGAL, but Manifold renders it, and creates an STL file that has no errors. It doesn't create an error-free 3MF.

adrianVmariano, 2025-02-13:
You **never** get an error from render when there is only one object because the rendering engine doesn't run.  To tell if an object is invalid you always have to have a second object.   I think I did see that manifold can handle objects that touch on edges, so that would explain why manifold works, though if the saved output may have errors that's not exactly a success story.

amatulic, 2025-03-04:
The action is to identify a case where non-triangulated faces are a problem, and if it isn't, then remove that from the validity check. I can do some experiments on my end.