BelfrySCAD/BOSL2 #1567: [BUG] WARNING: variable "fast" not specified as parameter

AlecMRogers, 2025-02-16:
**Describe the bug**
The bug results in both a warning and probably the lack of use of the parameter "fast", which is pretty essential to my call to vnf_hull()

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/polyhedra.scad>
include <BOSL2/vnf.scad>
vnf_polyhedron(vnf=vnf_hull( torus(d_maj=4, d_min=4) , fast=true ) );
```

**Expected behavior**
I was hoping for a speedup.

**Screenshots**
```
Parsing design (AST generation)...
Saved backup file: /Users/arogers/Documents/OpenSCAD/backups/Ball-backup-kvnNyPWp.scad
Compiling design (CSG Tree generation)...
[WARNING: variable "fast" not specified as parameter in file Ball.csg, line 9](https://github.com/BelfrySCAD/BOSL2/issues/9,/Users/arogers/Library/Mobile%20Documents/com~apple~CloudDocs/bits/projects/CAD/StringRings/output/Ball.csg)
Compiling design (CSG Products generation)...
Geometries in cache: 42
Geometry cache size in bytes: 72544
CGAL Polyhedrons in cache: 30
CGAL cache size in bytes: 0
```

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2025.01.27
- Other libraries used: BOSL2

AlecMRogers, 2025-02-16:
I looked at existing bugs, this might have something to do with attachment() refactoring?
In any case, my current algorithm calls vnf_hull( vnf_join([x,y,z]) ) , and halts due to complexity / buffer size / something else.

adrianVmariano, 2025-02-16:
The function form of vnf_hull() not not have a fast option.  It has to return a vnf and there is no "fast" method to do that.

adrianVmariano, 2025-02-16:
The docs do seem somewhat confused.

adrianVmariano, 2025-02-17:
Tried to clarify this stuff in the docs.  Doc update may not appear live for a while.

closed 2025-02-17 by adrianVmariano