BelfrySCAD/BOSL2 #1571: sweep, path_sweep nd similar modiles should take shapes as attachments instead of arguments

eternal-sorrow, 2025-02-21:
**Is your feature request related to a problem? Please describe.**
Not every module that create shapes can be called as a function. So passing such shapes to these modules is impossible.

**Describe the solution you'd like**
An ability to pass shapes as attachments (children) instead of passing them as arguments. For that creating new versions of such modules might be required in order to not break existing APIs.

**Describe alternatives you've considered**
A way to convert a shape created by a module into a  vector of coordinates for these modules, but I think it's probably impossible.

**Example Code**
Right now these modules work like this (random example from Wiki):
```
include <BOSL2/std.scad>
elliptic_arc = xscale(2, p=arc($fn=64,angle=[0,180], r=3));
path_sweep(rect(2), path3d(elliptic_arc), scale=[1,2]);
```
I want them to work like this instead:
```
include <BOSL2/std.scad>
elliptic_arc = xscale(2, p=arc($fn=64,angle=[0,180], r=3));
path_sweep(scale=[1,2]){
    rect(2);
    path3d(elliptic_arc);
}; 
```
So that I could do this for example:
```
include <BOSL2/std.scad>
elliptic_arc = xscale(2, p=arc($fn=64,angle=[0,180], r=3));
path_sweep(scale=[1,2]){
    round2d(r=1) rect(2);
    path3d(elliptic_arc);
}; 
```

**Additional context**
....

adrianVmariano, 2025-02-21:
Yeah, that would be nice but that is impossible in OpenSCAD.  There is no way to get geometry information from geometry objects.

closed 2025-02-21 by adrianVmariano

eternal-sorrow, 2025-02-21:
Really? That's too bad...

adrianVmariano, 2025-02-21:
Maybe someday:  https://github.com/openscad/openscad/pull/3956