BelfrySCAD/BOSL2 #1572: [BUG] tanh returns nan for some real inputs

mp4man, 2025-02-22:
**Describe the bug**
A clear and concise description of what the bug is.

tanh returns nan for some valid inputs

**Code To Reproduce Bug**
```
// Code goes here.
```
echo("check",730.192,tanh(730.192),752.692,tanh(752.692));

ECHO: "check", 730.192, nan, 752.692, nan

**Expected behavior**
A clear and concise description of what you expected to happen.

tanh(x) has a value in (-1,1) for any real input, x .
The mathematical value of tanh(x) for these inputs is very close to 1 
tanh(x)=\frac{e^{2x}-1}{e^{2x}+1}
and the call should return the value 1.

The existing code (see below) uses exp(x) and exp(-x) to calculate sinh(x) and cosh(x)
and then uses these to compute tanh(x) = sinh(x)/cosh(x)

For sufficiently large positive x,  exp(x) == inf               // eg x = 709.8
and for sufficiently small negative x, exp(-x) == inf      // eg x = -709.8
the existing tanh(x) code returns nan for such values.

For these cases tanh(x) should return +1 or -1.

**Screenshots**
n/a

**Additional context**
A suggested fix:

 sinh(x)/cosh(x) = (exp(x)-exp(-x))/(exp(x)+exp(-x))   // see existing code
                           = (exp(2x)-1)/(exp(2x)+1)                     // multiply both and bottom by exp(x)
let t = exp(2x), 
           as x -> +inf, t -> +inf
           as x -> -inf, t -> 0
in the limit as t -> inf    (t-1)/(t+1) -> 1
in the limit as t -> 0      (t-1)/(t+1) -> -1
when x is too large or too small for the existing code we should return these limiting values

one suggestion for a fix would be 
function tanh(x) =
      assert(is_finite(x), "The input must be a finite number.")
      let (e = exp(2*x) + 1) 
            e == inf  ?  1   :  (e-2)/e;

**Existing code**

function sinh(x) =
    assert(is_finite(x), "The input must be a finite number.")
    (exp(x)-exp(-x))/2;

function cosh(x) =
    assert(is_finite(x), "The input must be a finite number.")
    (exp(x)+exp(-x))/2;

function tanh(x) =
    assert(is_finite(x), "The input must be a finite number.")
    sinh(x)/cosh(x);

adrianVmariano, 2025-02-22:
Fix looks reasonable.  A remark: when those functions were added there was some pushback that nobody would ever use them or need them.

revarbat, 2025-02-23:
Minor vindication!

mp4man, 2025-02-23:
Many thanks for the speedy update.
On reflection I have a cleaner solution: 
  
  function tanh(x) =
    x > 0 ? -tanh(-x) 
          : let (e = exp(2*x)) (e-1)/(e+1);

This avoids the (mathematically offensive) use of **== inf**
  
Btw. I'm playing with tanh  as a blending function for isosurfaces.

adrianVmariano, 2025-02-23:
Actually your first version may appear less elegant, but it does run almost twice as fast, which might even matter for isosurfaces.  

We'd be interested in seeing how you're using isosurfaces, as it's a new feature and we don't know what it's good for.  :)

Also be aware that a revision is coming to isosurface.scad.

closed 2025-02-24 by adrianVmariano