BelfrySCAD/BOSL2 #1574: Provide capability to round corners on arc() or pie_slice()

dan-p3rry, 2025-02-23:
**Is your feature request related to a problem? Please describe.**
pie_slice() or arc(wedge=true) sometimes needs rounded corners

**Describe the solution you'd like**
Be able to specify inner_corner_radius and outer_corner_radius for arc() and pie_slice()

**Describe alternatives you've considered**
I tried rounding the corner with mask2d_roundover(), but that requires 90 degree corners

[image]

**Example Code**
Here is the solution, valid up to 180 degree wedge:

```
module rounded_arc(radius, inner_corner_radius, outer_corner_radius, angle) {

    angle_adj = asin(outer_corner_radius/(radius - outer_corner_radius));
    assert(angle - 2*angle_adj > 0, str("Corner_radius is too large for given radius and wedge angle"));
    radius_of_ctrpt = inner_corner_radius/sin(angle/2);

    pt1 = polar_to_xy(r=radius - outer_corner_radius, theta=angle_adj);
    pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle);
    pt3 = polar_to_xy(r=radius - outer_corner_radius, theta=angle - angle_adj);
       
    hull() {
        translate(pt1) circle(outer_corner_radius);
        translate(pt2) circle(inner_corner_radius);
        translate(pt3) circle(outer_corner_radius);
        arc(r=radius, angle=angle - 2*angle_adj, wedge=false, 
                start=angle_adj);
    }
}
```

**Additional context**
I took a look at contributing guidelines, but I don't have the background/experience to start messing with arc() and/or pie_slice().  Thanks!

adrianVmariano, 2025-02-23:
I'm not sure what you mean about mask_angle requiring 90 degrees.  It has a mask_angle parameter.  That said, it's not a perfect solution to this problem due to the fact that arcs are curved.   Probably a full solution would allow you to give two or three (when wedge=true) rounding parameters.   The arc function is somewhat complicated, so robust implementation may be tricky.  What happens if you have requested N points, for example?

amatulic, 2025-02-23:
Also, what should happen if the outer width of the pie slice is less than the diameter of the outside corners? Shrink the outside corners? Abort with an error? If shrinking, then the specification might be max_outer_corner_radius to imply that it can get smaller depending on the geometry.

Also I note that the inner corner center changes with respect to the pie center, depending on the angle of the wedge. What should happen when the inner corner interferes with the outer corner?

revarbat, 2025-02-23:
Should assert an error. Something like:assert(inner_corner_r+outer_corner_r <= r);-Revar

adrianVmariano, 2025-02-23:
If you request roundings that don't fit it should be an error.  That's how rect() and other functions that round multiple corners handle this.   The behavior isn't well-defined for the case where they interfere with each other.

dan-p3rry, 2025-02-24:
> Also, what should happen if the outer width of the pie slice is less than the diameter of the outside corners? Shrink the outside corners? Abort with an error? If shrinking, then the specification might be max_outer_corner_radius to imply that it can get smaller depending on the geometry.

Second line of code:  assert(angle - 2*angle_adj > 0, str("Corner_radius is too large for given radius and wedge angle"));  Granted, this assert statement needs clarification

> Also I note that the inner corner center changes with respect to the pie center, depending on the angle of the wedge. What should happen when the inner corner interferes with the outer corner?

rounded_arc(radius = 85, inner_corner_radius = 30, outer_corner_radius=10, angle = 60);

[image]

Can be caught by:
`    assert(radius_of_ctrpt + inner_corner_radius < radius, str("Inner corner radius is too large for given arc radius and angle"));
`

Considering the corner cases (no pun intended), the circles used in hull() probably should be half or even quarter circles oriented correctly.

adrianVmariano, 2025-02-24:
The way arc() works is by generating a list of points.  For the largest amount of flexibility this rounding feature would not use hull() but would trim back the arc and add a sequence of points for the rounding(s).

dan-p3rry, 2025-02-24:
This function uses no hull(). Also no error checking because I'm not sure the best way to check errors inside a function.

```
ra = rounded_arc(62.5, 5, 10, 60);
polygon(ra);

function rounded_arc(radius, inner_corner_radius, outer_corner_radius, angle) =
    let(
        angle_adj = asin(outer_corner_radius/(radius - outer_corner_radius)),
        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_opt = radius - outer_corner_radius,
        pt1 = polar_to_xy(r=radius - outer_corner_radius, theta=angle_adj),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=radius - outer_corner_radius, theta=angle - angle_adj)
    )
    concat(
        path2d(arc(cp=pt1, 
                points=[polar_to_xy(r=radius_of_opt, theta=0), polar_to_xy(r=radius, theta=angle_adj)])),
        path2d(arc(r=radius, angle=angle - 2*angle_adj, wedge=false, 
                start=angle_adj)),
        path2d(arc(cp=pt3, points=[polar_to_xy(r=radius, theta=angle-angle_adj), 
                                        polar_to_xy(r=radius_of_opt, theta=angle)])),
        path2d([polar_to_xy(r=radius_of_opt, theta=angle), polar_to_xy(r=radius_of_ctrpt, theta=angle)]),
        path2d(arc(cp=pt2, points=[polar_to_xy(r=radius_of_ctrpt, theta=angle), 
                                    polar_to_xy(r=radius_of_ctrpt, theta=0)])),
        path2d([polar_to_xy(r=radius_of_opt, theta=0), polar_to_xy(r=radius_of_ctrpt, theta=0)])        
    );

```

adrianVmariano, 2025-02-24:
FYI, the way I see this working would be integration into arc() as an optional `rounding=` argument the same way that rect() for example has a rounding argument.  

What's the right way to decide on the number of points to produce for the main arc body and the roundings---if user specificies n does the total output contain n points?  And how are they divided between the roundings and body?  

In the code above, why are you constantly calling path2d?   Is there a case of non-coincident collinear points?   I suppose this will happen with a 180 deg arc and the current code.

revarbat, 2025-02-24:
There's three corners that could have theoretically different rounding. Are you thinking rounding all three to the same radius if rounding= is given a scalar, and otherwise accepting a list of three scalars describing each corner?

revarbat, 2025-02-24:
My general solution for vertex counts on rounded corners is `ceil(segs(r)*arc_ang/360)`, though that may be overridden if given `n=`.

adrianVmariano, 2025-02-24:
That's what I was thinking, list of roundings for each corner, or scalar for all the same.  

If we want to folllow the $fn/$fa/$fs variables it seems like that's straight forward.  The tricky case is when the user provides n= which is supposed to always produce exactly n points.  So then the points need to be partitioned up among four arcs.  (Or three as noted below.)

There's actually another complication I hadn't thought of.  If wedge=false then the corners are completely different.  

[image]

adrianVmariano, 2025-02-24:
And would the first point on an arc with wedge=true be the center point of the rounding at the inner corner?  or the start (or end) of that rounding?

revarbat, 2025-02-24:
Maybe have `n=` be able to accept a list of vertex counts per corner?

revarbat, 2025-02-24:
Actually, wouldn't `n=` be identical to `$fn=` when given a scalar, since the path is closed?

adrianVmariano, 2025-02-24:
Not if you interpret $fn to mean every arc separately gets $fn*angle/360 points on it, but n as meaning that exact point count.  In the former case, the little roundings get lots of points.

adrianVmariano, 2025-04-11:
```
function round_arc(radius, rounding=0, angle) =
    let(
        rounding = force_list(rounding,3),
        inner_corner_radius = angle==180?0 : angle>180 ? -rounding[0] : rounding[0], 

        arc1_opt_radius = radius - rounding[1],
        arc1_angle = asin(rounding[1]/arc1_opt_radius),
        arc1_cut  = radius - arc1_opt_radius*cos(arc1_angle),

        arc2_opt_radius = radius - rounding[2],
        arc2_angle = asin(rounding[2]/arc2_opt_radius),
        arc2_cut  = radius - arc2_opt_radius*cos(arc2_angle),        

        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_ctrpt_edge = radius_of_ctrpt*cos(angle/2),

        pt1 = polar_to_xy(r=arc1_opt_radius, theta=arc1_angle),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=arc2_opt_radius, theta=angle - arc2_angle)
    )
    assert(arc1_angle + arc2_angle<=angle, "Roundings are too large: they interfere with each other on the arc")    
    assert(arc1_cut+radius_of_ctrpt_edge<=radius, "Roundings are too large: center rounding interferes with first corner")
    assert(arc2_cut+radius_of_ctrpt_edge<=radius, "Roundings are too large: center rounding interferes with second corner")    
    [
      each if (rounding[0]>0) arc(cp=pt2, 
                                   points=[polar_to_xy(r=radius_of_ctrpt_edge, theta=angle),     // origin corner curve
                                   polar_to_xy(r=radius_of_ctrpt_edge, theta=0)])
           else [[0,0]],                         
      each if (rounding[1]>0) arc(r=rounding[1],cp=pt1,angle=[-90,arc1_angle],endpoint=false),   // first corner
      each arc(r=radius, angle=[arc1_angle,angle - arc2_angle], endpoint=rounding[2]==0),        // main arc section
      each if (rounding[2]>0) arc(r=rounding[2],cp=pt3,  angle=[angle-arc2_angle, angle+90]),    // second corner
    ];
```

There's a reworking of the above to do most of what's needed.  There's still the completely different case of rounding when wedge=false.  @dan-p3rry Did you happen to work out the rounding details for that situation?  

Above code I think avoids duplicate vertices, handles 3 different roundings for the 3 corners, which can be zero.  The deficiencies are that it doesn't have support for a fixed `n` output point count, and I think it needs a reverse option for when the arc is going the other direction.  I don't think we can just reverse the path because it needs to start at the other side of the corner to be consistent. 

Another question is whether a rounding on a 180 deg corner is an error, or is just ignored.

adrianVmariano, 2025-04-11:
I have this written up with the outstanding questions:

* Is it an error to give a nonzero rounding on a 180 deg corner, or just ignored?
* How do I handle user specified `n` giving the exactly number of points desired?  How should they be split among the 2-4 arcs in the result?  
* Does it actually make sense to allow a vector value `n`?  This gets complicated when a user gives a rounding with some zero entries.  Does the length of n have to match the length of nonzero entries in rounding plus 1?  Or the user can specify zero for zero rounding arcs?  Or create repeated points?  Seems messy.  
* Can we round in the case where `wedge=false`?  Right now I made this an error.

adrianVmariano, 2025-04-12:
I made it so `n` gives the number of points per arc, which may not be the most sensible choice, but it's easy to implement and easy to explain and gives full control over the point count.   And I made a choice for 180 deg arcs.  

Last thing that might be nice would be the case of `wedge=false`.

dan-p3rry, 2025-04-12:
I'm OK either way with rounding for `wedge=false`.   Also, what do you think about this scenario: 

[image]

I don't know that there's a clean way to round the corners on the inner radius currently.  Perhaps the rounding argument could support negative values for the arc1 & arc2 corners?

adrianVmariano, 2025-04-12:
Regarding the wedge=false case, it's better to have everything work than to have things fail.  But in order to make it work, we need to know how to compute the roundings and at the moment I don't know how to do that.  

If you want to provide a patch that enables negative roundings on the outside I'll include it.  That could presumably make a rounding option on `ring()` a possibility.  

On option to consider for problems like this is using path_join, which is designed to round things that are curved:

```
include<BOSL2/rounding.scad>
stroke(path_join( [arc(r=20,angle=[10,100]), zrot(100,[[20,0],[10,0]]),
                    arc(r=10, angle=[100,10]), zrot(10,[[10,0],[20,0]])],
                   joint=3,relocate=false,closed=true),width=.3);
```

[image]

Actually Example 11 for path_join is exactly this case.  It also does the arcs with wedge=false.

dan-p3rry, 2025-04-12:
So we don't need rounded_arc()?  Just update the documentation for arc() to reference the path_join() example?

adrianVmariano, 2025-04-12:
I mean, rounded_arc is easier than using path_join, and if you insist on circular curves you're out of luck with path_join.  But adding rounding to arc() is a bit redundant.  On the other hand, it's a significant simplification compared to using path_join.  It's in the current posted version at this point so not going to disappear.  

I can't exactly update the manual for every shape that lacks integrated rounding to say "And BTW, if you want to round this, go look at path_join."  If you push that idea you end up with a list of 40 things for every function "And if you want to do X to this go look at function h()".    

I think the idea is more captured by #89 but alas, nobody has stepped up to write the Rounding tutorial yet.

dan-p3rry, 2025-04-12:
OK, I will work on the negative rounding for arc1 & arc2 corners then.  Good brain exercise.

adrianVmariano, 2025-04-12:
Code that is merged into the latest master is the current code.   The code posted in this chat is old.

dan-p3rry, 2025-04-12:
This code handles negative radii for the two outer corners.  But with a couple caveats.
1.  I based this code on your code from yesterday, which is broken for the special case angle=180.  The code I provided some weeks ago was able to handle angle=180.  This new code doesn't fix the bug.
2.  Too large values for corners 2 & 3 are not caught consistently by the assert statements:
3. 
4. 
5. 
6. , they fail before that with a non-helpful error message.  And ridiculously large negative corner radii are not caught at all, they go through.  (For negative corner radii, the CP of that radius keeps moving further away, so the resulting arc might be technically OK even for ridiculous numbers, I'm not sure.)

`function round_arc(radius, rounding=0, angle) =
    let(
        rounding = force_list(rounding,3),
        inner_corner_radius = angle==180?0 : angle>180 ? -rounding[0] : rounding[0], 

        arc1_opt_radius = radius - rounding[1],
        arc1_angle = asin(abs(rounding[1])/arc1_opt_radius),
        arc1_cut  = radius - arc1_opt_radius*cos(arc1_angle),

        arc2_opt_radius = radius - rounding[2],
        arc2_angle = asin(abs(rounding[2])/arc2_opt_radius),
        arc2_cut  = radius - arc2_opt_radius*cos(arc2_angle),        

        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_ctrpt_edge = radius_of_ctrpt*cos(angle/2),

        pt1 = polar_to_xy(r=arc1_opt_radius, theta=arc1_angle),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=arc2_opt_radius, theta=angle - arc2_angle)
    )
    assert(arc1_angle + arc2_angle<=angle, "Roundings are too large: they interfere with each other on the arc")    
    assert(arc1_cut+radius_of_ctrpt_edge<=radius, "Roundings are too large: center rounding interferes with first corner")
    assert(arc2_cut+radius_of_ctrpt_edge<=radius, "Roundings are too large: center rounding interferes with second corner")    
    [
      each if (rounding[0]>0) arc(cp=pt2, 
                                   points=[polar_to_xy(r=radius_of_ctrpt_edge, theta=angle),     // origin corner curve
                                   polar_to_xy(r=radius_of_ctrpt_edge, theta=0)])
           else [[0,0]],                         
      each if (rounding[1]>0) arc(r=rounding[1],cp=pt1,angle=[-90,arc1_angle],endpoint=false)   // first corner
        else if (rounding[1]<0) arc(r=abs(rounding[1]), cp=pt1, angle=-[90, 180 - arc1_angle], endpoint=false),
      each arc(r=radius, angle=[arc1_angle,angle - arc2_angle], endpoint=rounding[2]==0),        // main arc section
      each if (rounding[2]>0) arc(r=rounding[2],cp=pt3, angle=[angle-arc2_angle, angle+90])    // second corner
        else if (rounding[2]<0) arc(r=abs(rounding[2]), cp=pt3, angle=[180+(angle-arc2_angle), angle+90] ),
    ];
`

dan-p3rry, 2025-04-12:
This code is copied from master & modified for negative rounding.  Now works with negative angles, no 180 degree bug.

`function _rounded_arc(radius, rounding=0, angle, n) =
    assert(is_finite(angle) && angle>-360 && angle<360, "angle must be strictly between -360 and 360")
    assert(is_finite(rounding) || is_vector(rounding,3), "rounding must be a scalar or 3-vector")
    // assert(all_nonnegative(rounding), "rounding values must be nonnegative")
    let(
        rounding = force_list(rounding,3),
        dir = sign(angle),

//        inner_corner_radius = abs(angle)==180?0 : abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],
        inner_corner_radius = abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],        
        arc1_opt_radius = radius - rounding[1],
        arc2_opt_radius = radius - rounding[2],
        check = assert(rounding[1]<arc1_opt_radius, "rounding[1] is too big to fit")
                assert(rounding[2]<arc2_opt_radius, "rounding[2] is too big to fit"),
        arc1_angle = asin(abs(rounding[1])/arc1_opt_radius),
        arc2_angle = asin(abs(rounding[2])/arc2_opt_radius),
        arc1_cut  = radius - arc1_opt_radius*cos(arc1_angle),
        arc2_cut  = radius - arc2_opt_radius*cos(arc2_angle),
 
        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_ctrpt_edge = radius_of_ctrpt*cos(angle/2),
        pt1 = polar_to_xy(r=arc1_opt_radius, theta=dir*arc1_angle),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=arc2_opt_radius, theta=angle - dir*arc2_angle),
 
        edge_gap1=radius-arc1_cut-radius_of_ctrpt_edge,
        edge_gap2=radius-arc2_cut-radius_of_ctrpt_edge
    )
    assert(arc1_angle + arc2_angle<=abs(angle), "Roundings are too large: they interfere with each other on the arc")   
    assert(edge_gap1>=0, "Roundings are too large: center rounding (rounding[0]) interferes with first corner (rounding[1])")
    assert(edge_gap2>=0, "Roundings are too large: center rounding (rounding[0]) interferes with second corner (rounding[2])")   
    [
      each if (rounding[0]>0 && abs(angle)!=180)
                               arc(cp=pt2,
                                   points=[polar_to_xy(r=radius_of_ctrpt_edge, theta=angle),     // origin corner curve
                                   polar_to_xy(r=radius_of_ctrpt_edge, theta=0)],
                                   endpoint=edge_gap1!=0,n=n)
           else repeat([0,0],rounding[0]>0 && abs(angle)==180 && is_def(n) ? n : 1),                        
      each if (rounding[1]>0) arc(r=rounding[1],cp=pt1,angle=[-90*dir,dir*arc1_angle],endpoint=dir*arc1_angle==angle,n=n)           // first corner
        else if (rounding[1]<0) arc(r=abs(rounding[1]), cp=pt1, angle=-[dir*90, dir*180 - arc1_angle], endpoint=dir*arc1_angle==angle,n=n),
      each if (arc1_angle+arc2_angle<abs(angle))
                      arc(r=radius, angle=[dir*arc1_angle,angle - dir*arc2_angle], endpoint=rounding[2]==0,n=n),                 // main arc section
      each if (rounding[2]>0) arc(r=rounding[2],cp=pt3,  angle=[angle-dir*arc2_angle, angle+dir*90],endpoint=edge_gap2!=0,n=n)  // second corner
        else if (rounding[2]<0) arc(r=abs(rounding[2]), cp=pt3, angle=[dir*180+(angle-dir*arc2_angle), angle+dir*90],endpoint=edge_gap2!=0,n=n ),
    ];
`

adrianVmariano, 2025-04-12:
I thought I had carefully worked out the behavior for 180 deg.  What was the bug there?

adrianVmariano, 2025-04-12:
Can you repost your code en tirely in a code block?  It got kind of mangled.

adrianVmariano, 2025-04-12:
Note that I just double checked the angle=180 behavior and it seems precisely as desired.  

I am also not sure what you mean about the error handling.  When I tried:

`_rounded_arc(90, rounding=[5, 10, 115], 40)` 

on the existing code I got `[ERROR: Assertion '(rounding[2] < arc2_opt_radius)' failed: "rounding[2] is too big to fit" in file lib/BOSL2-fork/drawing.scad, line 906`

dan-p3rry, 2025-04-13:
> Note that I just double checked the angle=180 behavior and it seems precisely as desired.
>
> I am also not sure what you mean about the error handling. When I tried:
>
> `_rounded_arc(90, rounding=[5, [...]

Both the 180 deg. bug and the assertion error were in the code posted in this thread a couple days ago, not the code from the master.  So disregard.

dan-p3rry, 2025-04-13:
`
function _rounded_arc(radius, rounding=0, angle, n) =
    assert(is_finite(angle) && angle>-360 && angle<360, "angle must be strictly between -360 and 360")
    assert(is_finite(rounding) || is_vector(rounding,3), "rounding must be a scalar or 3-vector")
    // assert(all_nonnegative(rounding), "rounding values must be nonnegative")
    let(
        rounding = force_list(rounding,3),
        dir = sign(angle),

//        inner_corner_radius = abs(angle)==180?0 : abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],
        inner_corner_radius = abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],        
        arc1_opt_radius = radius - rounding[1],
        arc2_opt_radius = radius - rounding[2],
        check = assert(rounding[1]<arc1_opt_radius, "rounding[1] is too big to fit")
                assert(rounding[2]<arc2_opt_radius, "rounding[2] is too big to fit"),
        arc1_angle = asin(abs(rounding[1])/arc1_opt_radius),
        arc2_angle = asin(abs(rounding[2])/arc2_opt_radius),
        arc1_cut  = radius - arc1_opt_radius*cos(arc1_angle),
        arc2_cut  = radius - arc2_opt_radius*cos(arc2_angle),
 
        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_ctrpt_edge = radius_of_ctrpt*cos(angle/2),
        pt1 = polar_to_xy(r=arc1_opt_radius, theta=dir*arc1_angle),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=arc2_opt_radius, theta=angle - dir*arc2_angle),
 
        edge_gap1=radius-arc1_cut-radius_of_ctrpt_edge,
        edge_gap2=radius-arc2_cut-radius_of_ctrpt_edge
    )
    assert(arc1_angle + arc2_angle<=abs(angle), "Roundings are too large: they interfere with each other on the arc")   
    assert(edge_gap1>=0, "Roundings are too large: center rounding (rounding[0]) interferes with first corner (rounding[1])")
    assert(edge_gap2>=0, "Roundings are too large: center rounding (rounding[0]) interferes with second corner (rounding[2])")   
    [
      each if (rounding[0]>0 && abs(angle)!=180)
                               arc(cp=pt2,
                                   points=[polar_to_xy(r=radius_of_ctrpt_edge, theta=angle),     // origin corner curve
                                   polar_to_xy(r=radius_of_ctrpt_edge, theta=0)],
                                   endpoint=edge_gap1!=0,n=n)
           else repeat([0,0],rounding[0]>0 && abs(angle)==180 && is_def(n) ? n : 1),                        
      each if (rounding[1]>0) arc(r=rounding[1],cp=pt1,angle=[-90*dir,dir*arc1_angle],endpoint=dir*arc1_angle==angle,n=n)           // first corner
        else if (rounding[1]<0) arc(r=abs(rounding[1]), cp=pt1, angle=-[dir*90, dir*180 - arc1_angle], endpoint=dir*arc1_angle==angle,n=n),
      each if (arc1_angle+arc2_angle<abs(angle))
                      arc(r=radius, angle=[dir*arc1_angle,angle - dir*arc2_angle], endpoint=rounding[2]==0,n=n),                 // main arc section
      each if (rounding[2]>0) arc(r=rounding[2],cp=pt3,  angle=[angle-dir*arc2_angle, angle+dir*90],endpoint=edge_gap2!=0,n=n)  // second corner
        else if (rounding[2]<0) arc(r=abs(rounding[2]), cp=pt3, angle=[dir*180+(angle-dir*arc2_angle), angle+dir*90],endpoint=edge_gap2!=0,n=n ),
    ];
`

dan-p3rry, 2025-04-13:
```
function _rounded_arc(radius, rounding=0, angle, n) =
    assert(is_finite(angle) && angle>-360 && angle<360, "angle must be strictly between -360 and 360")
    assert(is_finite(rounding) || is_vector(rounding,3), "rounding must be a scalar or 3-vector")
    // assert(all_nonnegative(rounding), "rounding values must be nonnegative")
    let(
        rounding = force_list(rounding,3),
        dir = sign(angle),

//        inner_corner_radius = abs(angle)==180?0 : abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],
        inner_corner_radius = abs(angle)>180 ? -dir*rounding[0] : dir*rounding[0],        
        arc1_opt_radius = radius - rounding[1],
        arc2_opt_radius = radius - rounding[2],
        check = assert(rounding[1]<arc1_opt_radius, "rounding[1] is too big to fit")
                assert(rounding[2]<arc2_opt_radius, "rounding[2] is too big to fit"),
        arc1_angle = asin(abs(rounding[1])/arc1_opt_radius),
        arc2_angle = asin(abs(rounding[2])/arc2_opt_radius),
        arc1_cut = radius - arc1_opt_radius*cos(arc1_angle),
        arc2_cut = radius - arc2_opt_radius*cos(arc2_angle),
        radius_of_ctrpt = inner_corner_radius/sin(angle/2),
        radius_of_ctrpt_edge = radius_of_ctrpt*cos(angle/2),
        
        pt1 = polar_to_xy(r=arc1_opt_radius, theta=dir*arc1_angle),
        pt2 = polar_to_xy(r=radius_of_ctrpt, theta=0.5*angle),
        pt3 = polar_to_xy(r=arc2_opt_radius, theta=angle - dir*arc2_angle),
        edge_gap1=radius-arc1_cut-radius_of_ctrpt_edge,
        edge_gap2=radius-arc2_cut-radius_of_ctrpt_edge
    )
    assert(arc1_angle + arc2_angle<=abs(angle), "Roundings are too large: they interfere with each other on the arc")   
    assert(edge_gap1>=0, "Roundings are too large: center rounding (rounding[0]) interferes with first corner (rounding[1])")
    assert(edge_gap2>=0, "Roundings are too large: center rounding (rounding[0]) interferes with second corner (rounding[2])")   
    [
      each if (rounding[0]>0 && abs(angle)!=180)
                               arc(cp=pt2,
                                   points=[polar_to_xy(r=radius_of_ctrpt_edge, theta=angle),     // origin corner curve
                                   polar_to_xy(r=radius_of_ctrpt_edge, theta=0)],
                                   endpoint=edge_gap1!=0,n=n)
           else repeat([0,0],rounding[0]>0 && abs(angle)==180 && is_def(n) ? n : 1),                        
      each if (rounding[1]>0) arc(r=rounding[1],cp=pt1,angle=[-90*dir,dir*arc1_angle],endpoint=dir*arc1_angle==angle,n=n)           // first corner
        else if (rounding[1]<0) arc(r=abs(rounding[1]), cp=pt1, angle=-[dir*90, dir*180 - arc1_angle], endpoint=dir*arc1_angle==angle,n=n),
      each if (arc1_angle+arc2_angle<abs(angle))
                      arc(r=radius, angle=[dir*arc1_angle,angle - dir*arc2_angle], endpoint=rounding[2]==0,n=n),                 // main arc section
      each if (rounding[2]>0) arc(r=rounding[2],cp=pt3,  angle=[angle-dir*arc2_angle, angle+dir*90],endpoint=edge_gap2!=0,n=n)  // second corner
        else if (rounding[2]<0) arc(r=abs(rounding[2]), cp=pt3, angle=[dir*180+(angle-dir*arc2_angle), angle+dir*90],endpoint=edge_gap2!=0,n=n ),
    ];
```

adrianVmariano, 2025-04-13:
That's strange.  I've never had any problems embedding code.  I just type three back ticks, then start code on the next line, and finish code block with 3 backticks alone on a line.   Have never had that do anything strange.  

I have merged your revised code in.

closed 2025-04-13 by adrianVmariano