BelfrySCAD/BOSL2 #1583: Support for cycloidal gearboxes

therealadityashankar, 2025-03-01:
**Is your feature request related to a problem? Please describe.**
Cycloidal gearboxes are currently not supported via bosl2

**Describe the solution you'd like**
Support for cycloidal gearboxes

**Describe alternatives you've considered**
For now, I am sticking with planetary arrangements

**Example Code**
```
# add a mirrored gear for stability (defaults to true)
cycloidal_drive(params..., add_mirrored_gear=false)
```

**Additional context**

[image]

revarbat, 2025-03-01:
I remember futzing around with code to generate something like this. I'll have to dig around to see if I still have it.

therealadityashankar, 2025-03-02:
that would be awesome!

revarbat, 2025-03-02:
For reference: https://en.wikipedia.org/wiki/Cycloidal_drive