BelfrySCAD/BOSL2 #1586: right_triangle does not take all parameters

ebo, 2025-03-04:
I was playing with some of the BOSL2 primitives, and decided to use a right triangle.  The code reads:

// Example: Centered
//   right_triangle([60, 10, 40], center=true);
// Example: *Non*-Centered
//   right_triangle([60, 10, 40]);

When I give it the above examples, I get the following error:

[ERROR: Assertion 'is_vector(size, 2)' failed in file ../BOSL2/shapes2d.scad, line 904](https://github.com/BelfrySCAD/BOSL2/904,/home/ebo/.local/share/OpenSCAD/libraries/BOSL2/shapes2d.scad) 

**Code To Reproduce Bug**
```
right_triangle([60, 10, 40]);
```

**Expected behavior**
A right triangle of some height.

I was able to get it to work by only giving it 2 dimentions (ie right_triangle([60, 10]);)

adrianVmariano, 2025-03-04:
right_triangle is a 2d shape so it only accepts 2 dimensions.  Use linear_extrude.

ebo, 2025-03-04:
OK.  Thank you.  The internal documentation probably needs a minor 
tweak.

On Mar 4 2025 6:14 AM, adrianVmariano wrote:
> adrianVmariano left a comment (BelfrySCAD/BOSL2#1586)
> 
> right_triangle is a 2d shape so it only accepts 2 dimensions.  Use 
> linear_extrude.

ebo, 2025-03-04:
Just the fact that several parts of the internal documentation examples 
show 3D parameters, but other places they show only 2D + the error 
messages state you can only use 2D.  Minor thing, but it did confuse me, 
and cause me to post the message.

On Mar 4 2025 12:07 PM, adrianVmariano wrote:
> adrianVmariano left a comment (BelfrySCAD/BOSL2#1586)
> 
> What tweak do you recommend?  To me the documentation of this, which is 
> in shapes2d, seems oriented around 2d.  What would clarify it?

adrianVmariano, 2025-03-04:
You'll have to point to places in the "internal documentation" (?) that show 3D parameters to right_triangle.  I see no such examples anywhere.   The public facing documentation on the wiki is generated from the library itself so it's impossible for it to be wrong.

ebo, 2025-03-05:
I just got back to tracking this down.

Some place I had a note for shape3d.scad with the internal note 
"Example: Centered..." but that is for a wedge.  I must have somehow 
gotten right_triangle mixed up with the wedge as I was working and mixed 
it up.  That said, I literally would have sworn that I copy-pasted those 
lines of code from some place in my testing.

Sorry for the noise.  This should be closed.

On Mar 4 2025 5:58 PM, adrianVmariano wrote:
> adrianVmariano left a comment (BelfrySCAD/BOSL2#1586)
> 
> You'll have to point to places in the "internal documentation" (?) that 
> show 3D parameters to right_triangle.  I see no such examples anywhere. 
>   The public facing documentation on the wiki is generated from the 
> library itself so it's impossible for it to be wrong.

closed 2025-03-05 by adrianVmariano