BelfrySCAD/BOSL2 #1594: Allow single   isovalue   in function isosurface (which was previously an option)

mp4man, 2025-03-10:
**Is your feature request related to a problem? Please describe.**

I'm frustrated since I've been using this option in some code I'm developing

**  the function
```

function isosurface(f, isovalue, bounding_box, voxel_size, voxel_count=undef, reverse=false, closed=true, exact_bounds=false, show_stats=false, _mball=false) =
```

Now has the assertion

```
 assert(is_list(isovalue) && len(isovalue)==2 && is_num(isovalue[0]) && is_num(isovalue[1]), "\nIsovalue must be a range; use [minvalue,INF] or [-INF,maxvalue] for an unbounded range.")
```

**Describe the solution you'd like**

Please reinstate the single value option. I was using this to produce a surface that I could then thicken using an analytic gradient function.

adrianVmariano, 2025-03-10:
You need to use [c,INF] or [INF,c].  No functionality has been lost, but single isovalues turn out to be confusing because you have to pick one of those two interpretations.

adrianVmariano, 2025-03-10:
Note also that the previous behavior always turned c internally into [c,INF], so if it was working before, do that.   Which choice is correct depends on your function, though, which is why the single value option was removed.

amatulic, 2025-03-10:
Some history: When I first wrote the isosurface() function, I was thinking only in terms of metaballs, in which each metaball goes to infinity at its center, so a single isosurface value was allowed. Internally, this single value was converted to the range [isovalue, INF]. So if you want a single value like before, you'd use [isovalue, INF] -- which it was always using internally anyway. The metaballs() function (which calls isosurface() to generate the scene) does accept a single isovalue, but again, this is converted to the range [isovalue, INF].

Then we realized that there's a whole other class of isosurface that goes to infinity with distance away, requiring that isovalue be the *maximum* end of the range, as in [0, isovalue] or [-INF, isovalue].

Because there's no way to predict what sort of isosurface a user would be plotting, we decided to require a range as the isovalue input. As Adrian said, no functionality has been lost.

adrianVmariano, 2025-03-10:
Using a finite range probably wouldn't produce a uniformly thick shell.  If he computes the gradient and shifts the points he can ensure a uniform shell.

closed 2025-03-11 by adrianVmariano

mp4man, 2025-03-11:
Hi Adrian. That’s exactly right. Thanks.

mp4man, 2025-03-11:
I guess the two are just the two orientations of the same surface?

adrianVmariano, 2025-03-11:
Not sure what two you're referring to.

adrianVmariano, 2025-03-11:
It's in the (updated) docs---if your isosurface is bounded, one of the options will give you a bounded shape and the other choice will give you an unbounded shape with the bounded shape subtracted.  In practice this looks like a cube because the shape ends at the bounding box.