BelfrySCAD/BOSL2 #1597: [BUG] Cubetruss anchors incorrect

lozdown, 2025-03-17:
**Describe the bug**
The standard anchors created for a cubetruss are incorrect for any extents>1

**Code To Reproduce Bug**
```
cubetruss(size=10,extents=21, strut=2,  anchor=CENTER, spin=0, orient=UP) show_anchors();
```

**Expected behavior**
Anchors should be at corners and faces of cubetruss, however they become progressively further away from model as extents increase

**Screenshots**

[image]

**Additional context**

- OpenSCAD Version: 2021.01
- Other libraries used: BOSL2

adrianVmariano, 2025-03-17:
Appears maybe problem is setting `size` not setting `extents`.

adrianVmariano, 2025-03-18:
I merged a fix for this, but since I don't use this cubetruss stuff I'm not certain about it, so please let me know if everything works right now.

closed 2025-03-19 by adrianVmariano