BelfrySCAD/BOSL2 #1605: contour/metaballs docs

adrianVmariano, 2025-03-25:
- [ ] The 2d metaball animation is too fast. Maybe should be about half as fast as it runs now.
Maybe even a bit slower than that.
- [ ] You say "The metaballs module and function produces..." which is grammatically problematic.  I would accept "The metaballs module/function produces..."  or "module and function produce..."
- [ ] In the intro section you say "function literal accepting a 3-vector".  Just say "vector" to generically include 2d case.
- [ ] There is no grid units parameter.  Instead of writing "grid units" write "voxels (3d) and pixels (2d)" or something like that.  (Maybe not necessary, but consider if it's more clear in that case
- [ ] "The bounding box can also be specified as a scalar of a cube..."  Not sure what a scalar of a cube is.  Maybe "You can also specify the bounding box with a scalar, which defines a centered square or cube of the specified width".  Might also change the text "The bounding box parameter can be specified by" to "You can specify the bounding box by giving". 
- [ ] When introducing voxel/pixel size change "either of which can also be" to "which can be a scalar or vector size".  
- [ ] abiguity -> ambiguity
- [ ] Is there actually ambiguity about how to close the path?
I might just write 'open-ended contour path where some paths may end at the bounding box'
- [ ] In run time section, might say "...which can be long, especially in 3D"
- [ ] Instead of "8,000,000" might write "eight million"
- [ ] "The 3d spherical metaball and 2d circular metaball therefore has a..."  has->have
- [ ] Delete "The exact way that cutiff is defined geometrically varies for different ball types; see below for details".  Only torus has any mention of this and everything is fairly uniform.
- [ ] For metaballs2d, I would say "When called as a function, a region or list of paths is returned."  Not necessary to say it's a list of "one or more".
- [ ] I suggest in the usage message having a separate line for closed=false where instead of region= it says path_list=
- [ ] Do not compare to the 3D case.  So delete "As with the 3D metaballs."  
Assume the user reads this section of the manual first.
- [ ] Again, delete the thing about cutoff varying.  The only mention is for mb_ring
- [ ] in mb_trapzeoid change "bottoms width" to "bottom width"
And at the end of mb_trapezoid, "has faster" -->"have faster"
- [ ] For mb_stadium maybe "Set the size parameter to [width,height] to get an object that fits inside a rectangle of that size.  (You can also give a scalar size, which produces a circle.)"
- [ ] In the closed/unclosed paths section
Might title the section "The closed parameter"
- [ ] It doesn't make sense to say that a parameter is "unavailable" and then say it's "always true".  Instead describe what metaballs module does.  (Recover lost changes to this doc section.)
- [ ] Move "any 2d cross section is a contour" to the end of the paragraph as "Note that any 2d cross section..."
- [ ] isvalues should be ranges for 2d except the case of a function form with closed=false
- [ ] The section on the isovalue parameter for isosurfaces is confusing/wrong
Should say "function literal taking 3 parameters (x, y and z) for isosurface or two parameters (x and y) for contour..."
Note that users can always name parameters anything they want.  They aren't required to name them anything in particular in their function literal.  As written it sounds like they take vector parameters.
- [ ] Bounding box comments apply also to isosurface section

amatulic, 2025-03-25:
> I suggest in the usage message having a separate line for closed=false where instead of region= it says path_list=

How would that look in the code comments? Like this?
````
// Usage: As a function (when closed=true)
//   region = metaballs2d(spec, bounding_box, pixel_size, [isovalue=], [closed=true], [use_centers=], [smoothing=], [exact_bounds=], [show_stats=]);
// Usage: As a function (when closed=true)
//   path_list = metaballs2d(spec, bounding_box, pixel_size, [isovalue=], closed=false, [use_centers=], [smoothing=], [exact_bounds=], [show_stats=]);
````

adrianVmariano, 2025-03-25:
Could do it that way, or just one usage heading with closed=true on one and closed=false on the other.    Your way is probably a bit more clear, but maybe lose parens and write "as a function with closed=true"

You need more examples for contour.  Here are a few.  Also you need an example or examples showing the effect of smoothing, maybe overlaying a no smoothing output with a smoothing output?  

```
// f = function(x,y)sin(3*atan2(y,x))+norm([x,y]);
// contour(f,isovalue=2.5, bounding_box=8);

//f = function(x,y)(x^2 + y - 11)^2 + (x + y^2 - 7)^2;
//contour( f, bounding_box=12, isovalue=125, pixel_size=.5);

//f = function(x, y)  10 + x^2 - 10* cos(360*x) + y^2 - 10* cos(360* y);
//contour( f, bounding_box=10.24, isovalue=20, pixel_size=.2);  // changes a lot on pixel size

//f = function(x,y) sin(x)*sin(y) * sin(x+y);
//contour(f, bounding_box=720, isovalue=.1);

//f = function(X,Y) log(1 + X^2 + Y^2) * sin(X*180/PI) * sin(Y*180/PI);
//contour(f,bounding_box=12.5,isovalue=.2);

// Try other isovalues
// f = function(X,Y) -20 * exp(-0.2 * sqrt(0.5 * (X^2 + Y^2))) - exp(0.5 * (cos(360 * X) + cos(360 * Y))) + exp(1) + 20;
// contour(f,bounding_box=10,isovalue=8,pixel_size=.1);

// Cassini oval
//c=4;  a=2;
//f = function(x,y) (x^2 + y^2)^2 - 2*c^2*(x^2 - y^2);
//contour(f,bounding_box=14, isovalue=a^4);

// folium of descartes
// a=2;
// f = function(x,y) x^3+y^3-3*a*x*y;
// contour(f,isovalue=0, bounding_box=8);

amatulic, 2025-03-25:
Lost text about the closed parameter is from your comment here: https://github.com/BelfrySCAD/BOSL2/pull/1595#issuecomment-2743286038

adrianVmariano, 2025-03-25:
I'm puzzled by why you don't see a use case for supporting ranges.  Contour the module does not produce a curve.  It produces a region bounded by one or two curves.  Which part is the interior is determined by how the isvalue RANGE is specified.  And with the current implementation, it's wrong sometimes.  It's exactly the same as in 3D.  You are currently doing a range, you just have it fixed to [c,inf].  It's the same as with the 3d case.  Both cases can occur.  In my examples, half of them come out backwards.  So how is one supposed to deal with those backwards cases where I wanted [-inf,c] instead?  As I said, you need to answer the questions in the paragraph about how to work around the non-support of ranges in the docs if you aren't going to support them.  You need something like:

"Unlike the 3d isosurface, the contour module only accepts a scalar isovalue and always assumes the range [c,inf].  If you need it to be [-inf,c] instead then do X.  If you need it to be [c1,c2] do X."   That is, basically provide workarounds for the limitation, and call out that it's inconsistent with the 3d version, because a user will be surprised in switching between the two apparently equivalent modules that have this arbitrary API inconsistency.  

If there's some major programming challenge here maybe you can supply a "reverse" argument that flips which side is the inside?

adrianVmariano, 2025-03-26:
When I wrote "The section on the isovalue parameter for isosurfaces is confusing/wrong
Should say "function literal taking 3 parameters (x, y and z) for isosurface or two parameters (x and y) for contour..."
Note that users can always name parameters anything they want. They aren't required to name them anything in particular in their function literal. As written it sounds like they take vector parameters."

I meant the function parameter, not the isovalue parameter and was referrering to this text:

Parameter f (function): To provide a function, you supply a [function literal](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Function_literals) taking a 3D coordinate [x,y,z] (for isosurface()) or a 2D coordinate [x,y] (for contour()) as input to define the grid coordinate location and returning a single numerical value. You can also define an isosurface using an array of values instead of a function, in which case the isosurface is the set of points equal to the isovalue as interpolated from the array. The array indices are in the order [x][y][z] in 3D, and [x][y] in 2D.

See how it says that the function literals input to isosurface and contour should be vectors?  

I had a thought on the API for contours.  For the function form, it is clear that a scalar isovalue unambiguously makes sense, so this should be supported, and you can do that by just allowing either a scalar or range as input.  If the user gives a scalar, the `closed` argument is just NOT ALLOWED (or alternatively is ignored...though you know I don't prefer that), and you get the paths defined by f(x,z)=c.   If you instead give a range, then closed is allowed and works as it currently does.

adrianVmariano, 2025-03-26:
It occurred to me that the closed argument could be entirely dispensed with.  If you give a single isovalue then you return the path for that isovalue.  If given a range, you run as if closed=true so that it matches the behavior of the module.  The only case this doesn't cover would be a user who wants an isovalue at two places, but with the closed=false behavior.  But that's just two isovalues and could be run separately and concatted together.  I guess there would be a performance penalty....

amatulic, 2025-03-26:
> It occurred to me that the closed argument could be entirely dispensed with.

I recall commentinng in another PR that I didn't see the point of the closed argument. It makes sense for isosurfaces in case you want to tile a surface by joining the open ends. There's no need for open-ended paths in 2D though. If you want to tile polygons, you can just do that with union().

adrianVmariano, 2025-03-26:
I'm not sure what you mean about if you want to tile polygons.  Somebody may want just the path to use for something.  Maybe they want to extrude along it.    Maybe they want to attach it to some other path for some purpose.    In fact, the use case for open ended paths in 2D is appears much more clear than the use case for open surfaces in 3d, which I can't really think of any legitimate purpose for beyond visualization/debugging.  

You aren't everybody.  You don't know every possible use and application.  (I don't either.)  The only way you can legitimately say that there's no use for X is if you can easily produce X through some simple mechanism from what is available without it.

My argument that we can do without the closed argument basically boils down to the last case.  The following cases exist:

1. You get just the path(s) f(x,y)=c  (effectively the open case)
1. You get the path(s) bounding the region c1<=f(x,y)<=c2, where c1 or c2 may be infinite (effectively the closed case)
1. You get the union of the paths f(x,y)=c1 and f(x,y)=c2.  This is the open case with a path range [c1,c2]

I see no way that you can dismiss any of these as not useful.  It's nontrivial to convert between 1 and 2. so it seems clear that both should be produced.  The third one can be made from 1 by running twice, so it's possible to produce even if only 1 and 2 are available.  The downside is the runtime is double.  But as long as you have 1 and 2, everything is possible.  

So what I'm getting at here is that the closed=true vs closed=false behavior can be controlled by whether the user supplies a scalar or range for the isovalue instead of by a separate `closed` parameter.  As far as I can recall, you never articulated this idea.  It's not a change in what the program DOES, just a change in the API.

amatulic, 2025-03-28:
I believe I got the ranges working. A range is needed for some of your examples. Comments on the examples:

First one seems like something that could be generated directly, without an implicit polygon. Is there anything one can say about it other than it requires a range to display properly?

Cassini oval is exactly what you get with two metaballs. That deserves a mention. Maybe I could show them side by side.

Folium of Descartes: If someone actually wanted this, you can get an exact path with the parametric equation form, in either polar or cartesian coordinates, so why approximate it with a contour? Also, the contour doesn't cross.

adrianVmariano, 2025-03-28:
I don't think my examples needed finite ranges, though some of them might be interesting with finite ranges, like a field of rings?   But I certainly noticed that some were inverted with the [c,inf] behavior that was available.  

Yes, first one is meant to be a simple function that the user can look at and understand why it made that shape.  I think we have the sphere in that role for isosurface.  Could in fact do a circle as well for a simple example.  I wrote that example because the circle seemed so boring.  

Is cassini oval exactly equivalent to a a metaball or only qualitatively.  It doesn't appear to be the same as 1/r^2+1/(r-c)^2 mathematically.  

For folium of descartes, it seems interesting  that the contour doesn't cross or even meet.  Worth calling out this behavior, maybe, since the curve does include the intersection point.  Why is that point missing?  Can you provide conditions for when intersection points will or won't get included?   What about an example where they *do* meet?   Or is it the case that intersections always get split apart?  

I do agree that it's not the most exciting example, though.  It interesting in 3D:

[image]

It does seem a little funny to have to include the INF endpoint when I have set closed to false.  

```
ff = function(x,y,z) x^3+y^3-3*z*x*y;
isosurface(ff, bounding_box=8, isovalue=[0,INF], closed=false, voxel_size=.1);      
```

Speaking of contours crossing...or not...how about this:

```
g = function(x,y)
      let(
         theta=atan2(y,x),
         r = norm([x,y])
      )
      r*sin(3*theta-theta^2/20+40*r);
contour(g, bounding_box=[[-23,-13],[9,12.5]], isovalue=0, pixel_size=.2);
```

[image]

amatulic, 2025-03-28:
I get an error in path assembly with that last example.
Oddly, when I execute the steps individually in contour(), that is, call _contour_pixels first, then _contour_vertices, then _assemble_partial_paths, it works fine.

adrianVmariano, 2025-03-28:
Hmmm.  I did not get an error, so it must have some connection to your recent code revisions.  (I mean, could still be a path assembly bug...but I won't be able to replicate it.)

amatulic, 2025-03-28:
I fixed it. I had to change EPSILON to eps in assemble_partial_paths because EPSILON was too small for some test cases in which interpolation differences were ~1e-8; it's now testing against an `eps` argument that defaults to 1e-7. Why it worked when done in steps, and not when done in the contour() function, I still don't know.

amatulic, 2025-03-29:
Raised PR which should cover the points in this issue.
The last point in the first comment about not calling region() on an empty list, I wasn't sure what to do. I put an assert in there to make it abort if no contour paths are returned. It aborts either way, but at least now the user knows what happenened.

adrianVmariano, 2025-04-05:
You didn't like my contour example above?  

Are we ready to close this?

amatulic, 2025-04-05:
I didn't know that last one was intended to be used in an example. I thought you were just illustrating something interesting (it doesn't cross if you zoom way in). I also didn't know what to say about it in example commentary because wasn't sure of its educational value as an example. What do you suggest?

adrianVmariano, 2025-04-05:
Well, there's some educational value in just showing a diversity of things that can be done.  

In this case you also have something worth talking about regarding the non-intersection in the center.   That could be surprising and puzzling to people, so it seems worth mentioning.

amatulic, 2025-04-05:
Added in pull request #1618
You can close this issue.

closed 2025-04-05 by adrianVmariano