BelfrySCAD/BOSL2 #1611: Missing functions

alexgrca, 2025-03-30:
Hello all, I am quite new with openscad.
I want to make a simple fillet in a prism, something really easy.
I see that there is a fillet function in the library but I cannot found it.
It should be in mescellaneous, right??

I downloaded the zip file but I cannot find it there.
Can anybody help me?

Thanks a lot

RAMilewski, 2025-03-31:
fillet() is in shapes3d.scad.   See the [function index](https://github.com/BelfrySCAD/BOSL2/wiki/AlphaIndex) in the [wiki](https://github.com/BelfrySCAD/BOSL2/wiki)

adrianVmariano, 2025-04-06:
For the core functions and modules you don't need to know which file they are in.  You should be doing

include<BOSL2/std.scad>

after installing as described in the README.

closed 2025-04-06 by adrianVmariano