BelfrySCAD/BOSL2 #1637: [BUG] bent_cutout_mask misses volume in render with negative corners

pkriens, 2025-04-16:
The `bent_cutout_mask` renders differently in preview mode than render mode when using it with negative (outgoing) corners. 

I needed a cutout in a tube and used the mask to create a gentle rounded corner to go into the cutout, then nice rounded corners to go forward, up, and then another corner back to the edge of the tube.

[image]
[image]

This worked surprisingly well but when I later rendered it it turned out that the top of the cutout was not properly rendered.

[image]

Additionally, some remarks concerning the function:

* the mask is much larger than seems necessary. I had to work around it not cutting out other parts of my design. Could this be controlled?
* I guess it is a mask but it would be nice if it was attachable?

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

$fn=$preview ? $fn : 100;
    
cutout = subdivide_path(rect([40,30], rounding=[-5,-5,5,5]),100);
            
diff()            
tube( or=50, h=100, wall=4)
tag("remove") 
up(35) 
bent_cutout_mask(50-2, 4, cutout)            
;
```
**Expected behavior**
Identical behavior in render and preview.

**Additional context**
- OpenSCAD Version: 2025.03.19 git commit 2c9f43c00
- BOSL2: commit eecd3e02 (16 April 2025)

adrianVmariano, 2025-04-16:
It appears that the sharp corner (approaching zero degrees) leads to a long edge with no samples once offset() runs on it (internally to generate the mask).   I think this is in the category of cases that simply will not work.  You need to add an extra extension of your mask upwards (like is done in the slot, example 12).   I can add a note about this limitation to the docs.  

It appears that you found an OpenSCAD bug, by the way.  If you run your model on the stable version it fails with a rendering error, which is probably the correct behavior.  

Regarding the size of the mask, I chose the current approach because of simplicity.  I could intersect the mask with a tube (or you could).  This may make it slower, and it's already not fast.  It's conceivable that the outer surface could be redesigned with some significant effort to compute a planar projection instead of  using a simple point.   Also the inner section could probably be redone like join_prism works, but then the user must ensure that the overlap parameter is big enough.   I suppose intersecting with a tube could be a user option.  I also have not done any timing tests.  

For anchors, what anchors would you like?  I think the reason I didn't bother supporting anchors was that they seemed like a mechanism for putting the mask in the wrong place and kind of useless for putting the mask in the right place.   So how do you see this working?

adrianVmariano, 2025-04-24:
@pkriens Any comments on how you see anchoring working?  Or the idea of reworking the model to be less oversized?

pkriens, 2025-04-25:
On vacation (Corsica!) will react in about 4 weeks when I got my computer ...

pkriens, 2025-05-13:
[image]
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

$fn= $preview ? 30 : 200;

fudge=0.8;
r = 20;
profile_width = 6;
x = r-profile_width/2-fudge;
y = -sqrt( r^2 - x^2);

cyl(r=r,h=10)
let(a=parent())
move([r-5,-50,0])
cuboid([10,0.1,10])
prism_connector(
    circle(r=profile_width/2),
    a, [x,y,0],
    parent(), CTR,
    fillet=2,
    overlap=0,
    debug_pos=false
    )
    
;
```
This is the basic idea. If I could specify the align in the anchor it would be easier I think:
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

$fn= $preview ? 30 : 200;

r = 20;
profile_width = 6;

cyl(r=r,h=10)
let(a=parent())
move([r-5,-50,0])
cuboid([10,0.1,10])
prism_connector(
    circle(r=profile_width/2),
    a, [FRONT,LEFT],
    parent(), CTR,
    fillet=2
)    
;
```

I can see this is tricky with the angle of the connection and the fillet playing a crucial role.

adrianVmariano, 2025-05-13:
How is that connected to bent_cutout_mask?

pkriens, 2025-05-15:
My mistake, I confused this with the align feature on `prism_connector`. 

I was thinking along the lines of:
```
diff()            
tube( or=50, h=100, wall=4)
attach(FRONT,TOP /** ?? **/, align=TOP) 
bent_cutout_mask(50-2, 4, cutout) 
;
```

Is there a reason `tubes` have no anchors on the inside?

adrianVmariano, 2025-05-16:
So basically you are hoping that anchors for bent_cutout_mask can provide a way to align your cutout shape with the top edge of a cylinder instead of you having to design your mask there, or position it there.  It's a bit unclear to me that this capability is worth the trouble of adding attchment capability that will only work correctly for one anchor and will give garbage in every other case.   Do you think it's such a useful thing?  

I think nobody ever thought about putting anchors inside tubes.  The anchor system has no simple mechanism for this.  We could add named anchors, though it's a bit inflexible compared to the outside anchors.  And I'm not sure what they would be called.  Would 12 named anchors inside tubes be useful?   For a rectangular tube there are 24 locations, which is a lot of names.

pkriens, 2025-05-16:
> Do you think it's such a useful thing?

Don't know but my main objection to BOSL2 is the inconsistencies in the
API. I just tried to use tubes but then found out too late you could not
round the top and bottom edges. I assumed that bent_cutout_mask was
attachable because almost everything  is but then it did not work. It is
not a deal breaker (far from it) but it then takes time to go back to the
docs and read them more carefully. Especially now with restore() it would
be nice that all shapes at least repeat their children. But I guess I am
more of an API consistency freak than you are :-)

> I think nobody ever thought about putting anchors inside tubes. The
anchor system has no simple mechanism for this. We could add named anchors,
though it's a bit inflexible compared to the outside anchors. And I'm not
sure what they would be called. Would 12 named anchors inside tubes be
useful? For a rectangular tube there are 24 locations, which is a lot of
names.

Would it be an idea to give a parameter to parent(constituent), where the
constituent would be a name for the constituent? E.g. for a tube parent()
would be the default outer cylinder but parent("inner") would give you the
inner cylinder? This would prevent the anchor explosion. It might actually
also be useful if you could give variations. For example, rounded_prism is
marvelous but for attach you often want to use the 'unrounded' prism
because the rounding messes with the anchors in a way I often dislike.

This might be useful for other shapes? I often construct walled shapes
(tube, container, encapsulated) with rounded_prism because I can round
every edge. However, that gives me two rounded_prisms. One for the outer
side and the other for the inner side. I then tried to create scaffolds but
that wasn't particularly easy. If I could make a module that allows me to
let the clients after me get the inner and outer shapes that would be
awesome!

I guess it is too much work but just an idea ...

Kind regards,
     Peter Kriens

adrianVmariano, 2025-05-16:
I think technically bent_count_mask is attachable---it just doesn't take children.  But all of the anchors it provides are garbage because they are derived from the VNF of the mask.  Only the "origin" anchor is going to be useful for anything.  

I think before parent() and reset() the utility of children for something like this was very unclear, but I suppose I can see that it might be useful, so I can make it take children.   

I don't understand the problem with rounded ends of tubes.  They are supported.   Or maybe you meant rect_tube?

The idea of parent("inner") doesn't prevent the explosion of information, because how does parent() get the information that goes with "inner"?  It needs to be propagated from the parent!  So then that information needs to be placed into the geometry data structure somehow.   The way this sort of multiple case situation is currently handled is with anchor types.  So I could create an "inner" atype that would let you propagate interior anchors instead of exterior anchors.   At least in theory.  (If I do this the easy way the interior anchors will still point to the outside of the tube, which seems wrong.  It's possible flipping them to point the other way may break something somewhere.  I don't know.)   The problem with the anchor type mechanism is that you only get one anchor type at a time, so if you have instantiated a cylinder with type "outer" you can't use any inner anchors, and vise versa.  

I'm a little bit puzzled about your issue with rounded_prism anchors.  It's supposed to work (and is documented to work) that by default, rounding of the rounded prism has no effect on the anchors.  Unless you select one of the other anchor types.  

Can't you create a shape with something like 

```
diff()
rounded_prism( .... outside ...)  let(outside=parent())
tag_this("remove") rounded_prism( ... inside ..) let (inside=parent()) 
  <children have access to inside and outside now>
```

I thought of the idea of having $inside and $outside propagated, but I don't have a way to set them, it turns out, because the children are not the children of the actual object.   I suppose there may be a more cumbersome way to create the correct geometry so it can be propagated, so maybe this is possible.

adrianVmariano, 2025-05-19:
The latest update adds children support to bent_cutout_mask().

adrianVmariano, 2025-05-21:
If you observe minor API inconsistencies we may be able to fix them if you bring them to our attention before we freeze BOSL2.  The situation with rect_tube() is confusing because the same API is used for two different things: sometimes (as for rect)  you use `rounding` to control the rounding of the corners of a rectangle and sometimes (as for cyl) you use `rounding` to control rounding of the ends.  So rect_tube() is following the first convention instead of the second one.   I think in the case of rect_tube() we do not know a way to produce circular rounding of the ends that works generically, so we can't add that capability.  

The idea of providing alternate geometries is interesting.  It could replace the `atype` mechanism currently used with something more flexible.  But to work neatly would require essentially a complete rewrite of the whole anchoring system.   All the modules that use anchors would in principle need to have a selection mechanism to choose the desired variant.  It's not clear how that would work.  

The idea of passing of $ variables with alternate geometries has the appeal that it's something that can be done without a massive rewrite of anything.  So the likelihood of implementation is higher.

adrianVmariano, 2025-05-25:
I'm a little clear on what actions remain to resolve this issue.  I made bent_cutout_mask take children.  I haven't changed anything else.  I'm not sure that it's useful to add more anchoring support here since it's a lot of effort for what seems like negligible payback since the anchors will mostly give garbage.  Even the alignment idea is a little suspect because as noted in the discussion above, you probably need the mask to extend beyond the top of the tube you're cutting.   I'm not sure if there's a real need to change the shape of the mask to be less oversized, which is also a significant effort.  

Modifications to tube() are a completely separate matter.  Is it worth propagating an `$inside` geometry?  And would it be OK if this geometry still works like a regular cylinder, e.g. without **outward** pointing anchors?  That would be a little weird for trying to place objects inside the tube.   I suspect (but am not certain) that something will go wrong if the anchors point in the reverse of the direction of the anchor vector.  An alternative, but less flexible way to provide anchors for the inside of a tube would be with named anchors, but then alignment won't work.